BoschSensortec/BME680_driver #1: Calibration Data Indexing question

wjd14, 2017-07-19:
Can you please take a look at lines 308, 309, 310 of bme680_defs.h? I'm developing a python driver, and using your code as a reference, and the numbering on those lines seems a bit off. Perhaps I'm mistaken, but it seems odd that two parameters are assigned to the index 26 and index 24 is skipped entirely. Thanks.

vicatcu, 2017-07-20:
I Agree this is (almost certainly) an error, looking at how and where these offsets are used:
```
dev->calib.par_h2 = (uint16_t) (((uint16_t) coeff_array[BME680_H2_MSB_REG] << BME680_HUM_REG_SHIFT_VAL)
		        | ((coeff_array[BME680_H2_LSB_REG]) >> BME680_HUM_REG_SHIFT_VAL));
dev->calib.par_h1 = (uint16_t) (((uint16_t) coeff_array[BME680_H1_MSB_REG] << BME680_HUM_REG_SHIFT_VAL)
		        | (coeff_array[BME680_H1_LSB_REG] & BME680_BIT_H1_DATA_MSK));
```

Unfortunately, it's not clear what it should be without input from Bosch, but my naive guess would be:
```

#define BME680_H2_MSB_REG	(24)
#define BME680_H2_LSB_REG	(25)
#define BME680_H1_LSB_REG	(26)
```

BoschSensortec, 2017-07-20:
@wjd14 & @vicatcu ,

The macros are correct. par_H1 and parH2 are 12bits each. The lower 4 bits are stored together at index 26 while the higher 12 bits of parH1 is stored at index 27 and the higher 8 bits of parH2 are stored at index 25. index 24 is reserved.

You can hence observe in the code that the lower bits of H2 are shifted down by 4 while the lower bits of H1 are masked.

vicatcu, 2017-07-20:
@BoschSensortec thanks, that makes sense. However, just to clarify a bit further, the implication then is that coeff_array[24] is an unused / not-accessed field. Is that correct? When / where can we verify this in a datasheet?

wjd14, 2017-07-20:
@vicatcu I think that coeff_array[24] is just unused. I went back and looked through the definitions again, and there are actually a few that aren't used, such as 4, 10, 17, and 18. I haven't seen this in any datasheet though, just by looking through their code The datasheet does say that "not all reserved registers are depicted".

closed 2017-07-20 by wjd14

wjd14, 2017-07-20:
Thanks for the help Bosch. Looking forward to using this sensor.

BoschSensortec, 2017-07-20:
@wjd14 ,

We're happy to help. If you have any projects that you would like to share, do link them here.