BoschSensortec/BME680_driver #2: Datasheet Questoin

wjd14, 2017-07-24:


wjd14, 2017-07-24:
The datasheet was found on [this](https://www.bosch-sensortec.com/bst/products/all_products/bme680) webpage if that is useful.

BoschSensortec, 2017-07-25:
@wjd14 ,

Thank you for pointing that out. Page 24 is correct. The description of page 0 is incorrect in section 5.3.1.4. 
The code on the other hand also marks PAGE_0 and PAGE_1 incorrectly, although it is swapped at the point of use and hence functions as expected.

In short

| Page | Addresses | spi_mem_page<4> |
|:------:|:---------:|:--------:| 
| 0 | 0x80 - 0xff | 0(default) |
| 1 | 0x00 - 0x7f | 1 |

wjd14, 2017-07-25:
Thanks for the clarification. I was pretty sure I had it right, but I just wanted to confirm with you. Is there a better place to report little datasheet things like this? Are github issues fine, or is there another way? Thanks.

BoschSensortec, 2017-07-25:
@wjd14 ,

Ideally issues are reported via the [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form). For all code related issues, it would be recommended to use the Github issue tracker.

closed 2017-07-25 by wjd14