BoschSensortec/BME680_driver #3: Calibration question

wjd14, 2017-07-26:
Hello, one more question.

Line 134 of defs.h says that the length of the first group of calibration parameters is 25
``#define BME680_COEFF_ADDR1_LEN		UINT8_C(25)``

But this comment on line 70 of bme680.c says the first group of cal parameters only has a length of 24.
``@note Registers 89h  to A1h for calibration data 1 to 24 from bit 0 to 7``

So where does the first group of calibration parameters begin and end? 
Thanks for all of the help.

BoschSensortec, 2017-07-28:
@wjd14 ,

There are reserved bits that are also read although the extracted values in the current release are the only ones mentioned in the comment. If you observed line 289 of bme680_defs.h, you will see that the least index is 1 and not 0. The whole space is 0x89 - 0xA2 -> 25 bytes.

wjd14, 2017-07-28:
Is the data read at 0x89 index 0 or 1?

wjd14, 2017-07-31:
Just tested it again, think I'm finally reading the calibration data correctly. Thanks for all of your help.

closed 2017-07-31 by wjd14