BoschSensortec/BME680_driver #4: Strange readings

correa, 2017-09-28:
Temperature readings are above the expected by a few degrees, humidity is constant 100% and pressure around 649 hPA. Gas never returns a result. I'm using atmega32, any ideas what I could try to debug this?

BoschSensortec, 2017-09-29:
@correa,

The temperature being slightly higher than ambient temperature is normal. This is due to the influence of the heating element of the gas sensor. For a periodic heating cycle, you could subtract an offset to know the ambient temperature.

Can you attach your project here so we can review the code? It seems that you haven't linked the right read and write functions.

Meroryan, 2017-10-10:
Hi there,
I have also the problem with the gas reading.
I use the example code 1 to 1 from the readme.
Never goes true:
if(data.status & BME680_HEAT_STAB_MSK)
			printf(", G: %d ohms", data.gas_resistance);

BoschSensortec, 2017-10-11:
Hi @Meroryan ,

Could you leave the read, write and delay function implementations in the comments here below and also a brief description of the platform that you are using?

Meroryan, 2017-10-11:
Hi @BoschSensortec,
my platform is a custom board with nordic nRF52832. BME680 is connected through I2C und set to addr 0x77. All GND connections can be switched off by µC with GPIO and FET.

`
void user_delay_ms(uint32_t period)
{
	nrf_delay_ms(period);
}

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

		TransferPending = 1;
		rslt = nrf_drv_twi_tx(&m_twi, BME680_ADDR, &reg_addr, 1, true);
		while ( TransferPending == 1) {}
		TransferPending = 1;
		rslt = nrf_drv_twi_rx(&m_twi, BME680_ADDR, reg_data, len);
		while ( TransferPending == 1) {}
			
    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
    uint8_t	send_tmp[200]	= {0};
		send_tmp[0] 	= reg_addr;
		memcpy(send_tmp +1, reg_data, len);

		TransferPending = 1;
		rslt = nrf_drv_twi_tx(&m_twi, BME680_ADDR, send_tmp, len +1, false);
		while (TransferPending == 1) {}
	
    return rslt;
}
`

BoschSensortec, 2017-10-11:
@Meroryan ,

Within the modified functions, could you create a run-time communication log? We can debug the log and get back to you with the root-cause of the issue. Also, if the return of any of the APIs in non-zero, print that too.

``` c
void user_delay_ms(uint32_t period)
{
    /* Print: DL <period> */
    nrf_delay_ms(period);
}

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
    
    TransferPending = 1;
    rslt = nrf_drv_twi_tx(&m_twi, BME680_ADDR, &reg_addr, 1, true);
    while ( TransferPending == 1) {}
    
    TransferPending = 1;
    rslt = nrf_drv_twi_rx(&m_twi, BME680_ADDR, reg_data, len);
    while ( TransferPending == 1) {}

    /* Print: WR <reg_addr> <len> <reg_data... > */

    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
    uint8_t	send_tmp[200]	= {0};
    send_tmp[0] 	= reg_addr;
    memcpy(send_tmp +1, reg_data, len);

    /* Print: RD <reg_addr> <len> <reg_data... > */

    TransferPending = 1;
    rslt = nrf_drv_twi_tx(&m_twi, BME680_ADDR, send_tmp, len +1, false);
    while (TransferPending == 1) {}
	
    return rslt;
}
```

Meroryan, 2017-10-11:
New functions:
```
void user_delay_ms(uint32_t period)
{
	NRF_LOG_INFO("DL: %dms", period);
	nrf_delay_ms(period);
}

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

		TransferPending = 1;
		rslt = nrf_drv_twi_tx(&m_twi, BME680_ADDR, &reg_addr, 1, true);
		while ( TransferPending == 1) {}
		TransferPending = 1;
		rslt = nrf_drv_twi_rx(&m_twi, BME680_ADDR, reg_data, len);
		while ( TransferPending == 1) {}
		
		//Print: RD <reg_addr> <len> <reg_data... >
		NRF_LOG_INFO(" RD: reg_addr: 0x%x,; length: %d,  ",reg_addr, len);
		for ( uint16_t i = 0; i < len; i++)
			{
				NRF_LOG_INFO("RD: reg_data[%d]: 0x%x", i, reg_data[i]);
			}
				
    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
    uint8_t	send_tmp[200]	= {0};
		send_tmp[0] 	= reg_addr;
		memcpy(send_tmp +1, reg_data, len);

		//Print: WR <reg_addr> <len> <reg_data... > 
		NRF_LOG_INFO(" WR: reg_addr: 0x%x,; length: %d,  ",reg_addr, len);
		for ( uint16_t i = 0; i < len; i++)
			{
				NRF_LOG_INFO("WR: reg_data[%d]: 0x%x", i, reg_data[i]);
			}
		
		TransferPending = 1;
		rslt = nrf_drv_twi_tx(&m_twi, BME680_ADDR, send_tmp, len +1, false);
		while (TransferPending == 1) {}
	
    return rslt;
}
```
```
LogOutput:
<info> app: Blinky example started.
<info> app: RD: reg_addr: 0x0,; length: 1,
<info> app: 	RD: reg_data[0]: 0x2E

<info> app: RD: reg_addr: 0x4,; length: 1,
<info> app: 	RD: reg_data[0]: 0xF3

<info> app: WR: reg_addr: 0x5A,; length: 3,
<info> app: 	WR: reg_data[0]: 0x71
<info> app: 	WR: reg_data[1]: 0x64
<info> app: 	WR: reg_data[2]: 0x65

<info> app: RD: reg_addr: 0x74,; length: 1,
<info> app: 	RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x75,; length: 1,
<info> app: 	RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x74,; length: 1,
<info> app: 	RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x72,; length: 1,
<info> app: 	RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x71,; length: 1,
<info> app: 	RD: reg_data[0]: 0x0

<info> app: WR: reg_addr: 0x75,; length: 7,
<info> app: 	WR: reg_data[0]: 0x8
<info> app: 	WR: reg_data[1]: 0x74
<info> app: 	WR: reg_data[2]: 0x8C
<info> app: 	WR: reg_data[3]: 0x72
<info> app: 	WR: reg_data[4]: 0x2
<info> app: 	WR: reg_data[5]: 0x71
<info> app: 	WR: reg_data[6]: 0x10

<info> app: RD: reg_addr: 0x74,; length: 1,
<info> app: 	RD: reg_data[0]: 0x8C

<info> app: WR: reg_addr: 0x74,; length: 1,
<info> app: 	WR: reg_data[0]: 0x8D

<info> app: DL: 173ms

<info> app: RD: reg_addr: 0x1D,; length: 15,
<info> app: 	RD: reg_data[0]: 0x20
<info> app: 	RD: reg_data[1]: 0x0
<info> app: 	RD: reg_data[2]: 0x80
<info> app: 	RD: reg_data[3]: 0x0
<info> app: 	RD: reg_data[4]: 0x0
<info> app: 	RD: reg_data[5]: 0x80
<info> app: 	RD: reg_data[6]: 0x0
<info> app: 	RD: reg_data[7]: 0x0
<info> app: 	RD: reg_data[8]: 0x80
<info> app: 	RD: reg_data[9]: 0x0
<info> app: 	RD: reg_data[10]: 0x80

<info> app:	RD: reg_data[11]: 0x0
<info> app: 	RD: reg_data[12]: 0x0
<info> app: 	RD: reg_data[13]: 0x0
<info> app: 	RD: reg_data[14]: 0x4

<info> app: DL: 10ms

<info> app: RD: reg_addr: 0x1D,; length: 15,
<info> app: 	RD: reg_data[0]: 0x80
<info> app: 	RD: reg_data[1]: 0x0
<info> app: 	RD: reg_data[2]: 0x59
<info> app: 	RD: reg_data[3]: 0xB9
<info> app: 	RD: reg_data[4]: 0x80
<info> app: 	RD: reg_data[5]: 0x79
<info> app: 	RD: reg_data[6]: 0x40
<info> app: 	RD: reg_data[7]: 0xA0
<info> app: 	RD: reg_data[8]: 0x50
<info> app: 	RD: reg_data[9]: 0xB1
<info> app: 	RD: reg_data[10]: 0x80
<info> app: 	RD: reg_data[11]: 0x0
<info> app: 	RD: reg_data[12]: 0x0
<info> app: 	RD: reg_data[13]: 0xA5
<info> app: 	RD: reg_data[14]: 0xE7

```

Mero

BoschSensortec, 2017-10-11:
@Meroryan ,

It seems like you are missing the initialization sequence.

``` c
        struct bme680_dev gas_sensor;

	/* You may assign a chip select identifier to be handled later */
	gas_sensor.dev_id = 0;
	gas_sensor.intf = BME680_SPI_INTF;
	gas_sensor.read = user_spi_read;
	gas_sensor.write = user_spi_write;
	gas_sensor.delay_ms = user_delay_ms;

	int8_t rslt = BME680_OK;
	rslt = bme680_init(&gas_sensor);
```

Meroryan, 2017-10-11:
I'm also a bit surprised about the log.
Every 60s I call:
```
struct bme680_field_data bme680_data;

void bme680_read(void)
{
	nrf_gpio_pin_set(BME680_EN_PIN);
	nrf_delay_ms(1);
	struct bme680_dev gas_sensor;
	uint8_t set_required_settings;
	uint16_t meas_period;
	
	gas_sensor.dev_id = BME680_I2C_ADDR_SECONDARY;
	gas_sensor.intf = BME680_I2C_INTF;
	gas_sensor.read = user_i2c_read;
	gas_sensor.write = user_i2c_write;
	gas_sensor.delay_ms = user_delay_ms;

	bme680_init(&gas_sensor);	

	/* Set the temperature, pressure and humidity settings */
	gas_sensor.tph_sett.os_hum = BME680_OS_2X;
	gas_sensor.tph_sett.os_pres = BME680_OS_4X;
	gas_sensor.tph_sett.os_temp = BME680_OS_8X;
	gas_sensor.tph_sett.filter = BME680_FILTER_SIZE_3;

	/* Set the remaining gas sensor settings and link the heating profile */
	gas_sensor.gas_sett.run_gas = BME680_ENABLE_GAS_MEAS;
	/* Create a ramp heat waveform in 3 steps */
	gas_sensor.gas_sett.heatr_temp = 320; /* degree Celsius */
	gas_sensor.gas_sett.heatr_dur = 150; /* milliseconds */

	/* Select the power mode */
	/* Must be set before writing the sensor configuration */
	gas_sensor.power_mode = BME680_FORCED_MODE; 

	/* Set the required sensor settings needed */
	set_required_settings = BME680_OST_SEL | BME680_OSP_SEL | BME680_OSH_SEL | BME680_FILTER_SEL | BME680_GAS_SENSOR_SEL;
		
	/* Set the desired sensor configuration */
	bme680_set_sensor_settings(set_required_settings,&gas_sensor);

	/* Set the power mode */
	bme680_set_sensor_mode(&gas_sensor);

	/* Get the total measurement duration so as to sleep or wait till the
	 * measurement is complete */

	bme680_get_profile_dur(&meas_period, &gas_sensor);
	user_delay_ms(meas_period); /* Delay till the measurement is ready */
	

	bme680_get_sensor_data(&bme680_data, &gas_sensor);

	//NRF_LOG_INFO("T: %.2d degC, P: %.2d hPa, H %.2d %%rH, G: %d ohms ", (uint32_t)(bme680_data.temperature), (uint32_t)(bme680_data.pressure), (uint32_t)(bme680_data.humidity), (uint32_t)bme680_data.gas_resistance);
	if(bme680_data.status & BME680_HEAT_STAB_MSK)
			NRF_LOG_INFO(", G: %d ohms", bme680_data.gas_resistance);
	
	nrf_gpio_pin_clear(BME680_EN_PIN);
}
```
Perhabs the NRF_LOG swallows something.
I make a log with my logic analyzer.

BoschSensortec, 2017-10-11:
@Meroryan ,

Can you also process the return values of the API? If non-zero, print the value.

Meroryan, 2017-10-11:
I changed buffer size for logging.
And add something like:
	if ( rslt != 0x00 )
	{
		NRF_LOG_INFO("bme680_init(&gas_sensor); return: %d", rslt);
	}
to all API calls. Always return with BME680_OK.
The new log output is:
```
<info> app: WR: reg_addr: 0xE0,; length: 1,
<info> app: WR: reg_data[0]: 0xB6

<info> app: DL: 10ms

<info> app: RD: reg_addr: 0xD0,; length: 1,
<info> app: RD: reg_data[0]: 0x61

<info> app: RD: reg_addr: 0x89,; length: 25,
<info> app: RD: reg_data[0]: 0x0
<info> app: RD: reg_data[1]: 0x9F
<info> app: RD: reg_data[2]: 0x66
<info> app: RD: reg_data[3]: 0x3
<info> app: RD: reg_data[4]: 0x3F
<info> app: RD: reg_data[5]: 0x5
<info> app: RD: reg_data[6]: 0x8F
<info> app: RD: reg_data[7]: 0x9
<info> app: RD: reg_data[8]: 0xD7
<info> app: RD: reg_data[9]: 0x58
<info> app: RD: reg_data[10]: 0xFF
<info> app: RD: reg_data[11]: 0x39
<info> app: RD: reg_data[12]: 0x1A
<info> app: RD: reg_data[13]: 0xE1
<info> app: RD: reg_data[14]: 0xFF
<info> app: RD: reg_data[15]: 0x2C
<info> app: RD: reg_data[16]: 0x1E
<info> app: RD: reg_data[17]: 0x0
<info> app: RD: reg_data[18]: 0x0
<info> app: RD: reg_data[19]: 0x6E
<info> app: RD: reg_data[20]: 0xF4
<info> app: RD: reg_data[21]: 0x5D
<info> app: RD: reg_data[22]: 0xF6
<info> app: RD: reg_data[23]: 0x1E
<info> app: RD: reg_data[24]: 0x2

<info> app: RD: reg_addr: 0xE1,; length: 16,
<info> app: RD: reg_data[0]: 0x3F
<info> app: RD: reg_data[1]: 0x8B
<info> app: RD: reg_data[2]: 0x2E
<info> app: RD: reg_data[3]: 0x0
<info> app: RD: reg_data[4]: 0x2D
<info> app: RD: reg_data[5]: 0x14
<info> app: RD: reg_data[6]: 0x78
<info> app: RD: reg_data[7]: 0x9C
<info> app: RD: reg_data[8]: 0x77
<info> app: RD: reg_data[9]: 0x66
<info> app: RD: reg_data[10]: 0x88
<info> app: RD: reg_data[11]: 0xC1
<info> app: RD: reg_data[12]: 0xE8
<info> app: RD: reg_data[13]: 0x12
<info> app: RD: reg_data[14]: 0xF3
<info> app: RD: reg_data[15]: 0x0

<info> app: RD: reg_addr: 0x2,; length: 1,
<info> app: RD: reg_data[0]: 0x16

<info> app: RD: reg_addr: 0x0,; length: 1,
<info> app: RD: reg_data[0]: 0x2E

<info> app: RD: reg_addr: 0x4,; length: 1,
<info> app: RD: reg_data[0]: 0xF3

<info> app: WR: reg_addr: 0x5A,; length: 3,
<info> app: WR: reg_data[0]: 0x71
<info> app: WR: reg_data[1]: 0x64
<info> app: WR: reg_data[2]: 0x65

<info> app: RD: reg_addr: 0x74,; length: 1,
<info> app: RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x75,; length: 1,
<info> app: RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x74,; length: 1,
<info> app: RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x72,; length: 1,
<info> app: RD: reg_data[0]: 0x0

<info> app: RD: reg_addr: 0x71,; length: 1,
<info> app: RD: reg_data[0]: 0x0

<info> app: WR: reg_addr: 0x75,; length: 7,
<info> app: WR: reg_data[0]: 0x8
<info> app: WR: reg_data[1]: 0x74
<info> app: WR: reg_data[2]: 0x8C
<info> app: WR: reg_data[3]: 0x72
<info> app: WR: reg_data[4]: 0x2
<info> app: WR: reg_data[5]: 0x71
<info> app: WR: reg_data[6]: 0x10

<info> app: RD: reg_addr: 0x74,; length: 1,
<info> app: RD: reg_data[0]: 0x8C

<info> app: WR: reg_addr: 0x74,; length: 1,
<info> app: WR: reg_data[0]: 0x8D

<info> app: DL: 173ms

<info> app: RD: reg_addr: 0x1D,; length: 15,
<info> app: RD: reg_data[0]: 0x20
<info> app: RD: reg_data[1]: 0x0
<info> app: RD: reg_data[2]: 0x80
<info> app: RD: reg_data[3]: 0x0
<info> app: RD: reg_data[4]: 0x0
<info> app: RD: reg_data[5]: 0x80
<info> app: RD: reg_data[6]: 0x0
<info> app: RD: reg_data[7]: 0x0
<info> app: RD: reg_data[8]: 0x80
<info> app: RD: reg_data[9]: 0x0
<info> app: RD: reg_data[10]: 0x80
<info> app: RD: reg_data[11]: 0x0
<info> app: RD: reg_data[12]: 0x0
<info> app: RD: reg_data[13]: 0x0
<info> app: RD: reg_data[14]: 0x4

<info> app: DL: 10ms

<info> app: RD: reg_addr: 0x1D,; length: 15,
<info> app: RD: reg_data[0]: 0x20
<info> app: RD: reg_data[1]: 0x0
<info> app: RD: reg_data[2]: 0x80
<info> app: RD: reg_data[3]: 0x0
<info> app: RD: reg_data[4]: 0x0
<info> app: RD: reg_data[5]: 0x80
<info> app: RD: reg_data[6]: 0x0
<info> app: RD: reg_data[7]: 0x0
<info> app: RD: reg_data[8]: 0x80
<info> app: RD: reg_data[9]: 0x0
<info> app: RD: reg_data[10]: 0x80
<info> app: RD: reg_data[11]: 0x0
<info> app: RD: reg_data[12]: 0x0
<info> app: RD: reg_data[13]: 0x0
<info> app: RD: reg_data[14]: 0x4

<info> app: DL: 10ms

<info> app: RD: reg_addr: 0x1D,; length: 15,
<info> app: RD: reg_data[0]: 0x80
<info> app: RD: reg_data[1]: 0x0
<info> app: RD: reg_data[2]: 0x59
<info> app: RD: reg_data[3]: 0x5F
<info> app: RD: reg_data[4]: 0xC0
<info> app: RD: reg_data[5]: 0x76
<info> app: RD: reg_data[6]: 0x18
<info> app: RD: reg_data[7]: 0x0
<info> app: RD: reg_data[8]: 0x57
<info> app: RD: reg_data[9]: 0x7
<info> app: RD: reg_data[10]: 0x80
<info> app: RD: reg_data[11]: 0x0
<info> app: RD: reg_data[12]: 0x0
<info> app: RD: reg_data[13]: 0xC5
<info> app: RD: reg_data[14]: 0x67

```

BoschSensortec, 2017-10-12:
@Meroryan , 

It seems that you are getting sensor data, You should expect
Temperature = 2587 = 25.87°C
Pressure = 97544 = 97.544 hPa
Humidity = 38373 = 38.373 %rH
Gas resistance = 94663 ohms

It is also observed that the heater stability bit is 0 which would happen for the first 2-3 periodic measurements after which you should expect the heater to be stable.

Meroryan, 2017-10-12:
Could my measurement sequence cause the problem?:
Every 60s ( later ) every 5min MCU wake up by wathdog reset >> init MCU >> power on the BME680 ( switch BME GND to Board GND with MOSFET >> call BME680_read() ( init > setup > sleep > measure ) >> power off the BME680 >> send Values over Bluetooth LE >> send MCU to deep sleep (system off mode ) >> wait for watchdog reset...

BoschSensortec, 2017-10-12:
@Meroryan ,

Once initialized, there is no need to reconfigure the sensor. There is typically no need to use a MOSFET to power the BME680 as the sleep currents are typically around 150nA (lower or same to leakage currents of typical MOSFETS). Also during the heat-up phase, the BME680 will draw a higher current, up-to 12mA. It could happen that since there is not enough power available to power the heater, a stable desired heater temperature is not reached in the 150ms of the heating duration. In case the MOSFET is crucial, it would be recommended to increase the capacitance for VDD. 

Although not explicitly mentioned in the README of the sensor API, a shorter duration like 150ms is coupled with a reading frequency of around 0.25Hz or higher. For a longer sleep duration, it is recommended to use the heater for a longer duration so as to get a stable heating temperature. For a sleep of 5 minutes, a recommended duration would be 1s. 

Regarding operation of the MCU, it is recommended to configure the sensor once at the beginning, 
Trigger forced mode > sleep > read sensor data >stream sensor data, provided the sensor is not disabled while the system is in sleep. This would also reduce the amount of communication required with the sensor effectively reducing active time of the MCU and hence reducing overall power consumption.

Meroryan, 2017-10-19:
@BoschSensortec
Thanks for the good support.
Next Step is to get BSEC lib up and running.

Meroryan, 2017-10-20:
Is there a special place to ask questions about BSEC lib?
I integrate the Cortex_M4F Version and the provided example.
Timing seems ok.
[image]

I print out the values provided in void output_ready(...):
```
<info> app: systick:259
<info> app: timestamp:6000
<info> app: iaq:0.00                                           << 0?
<info> app: iaq_accuracy:0                                << 0?
<info> app: temperature:25.81                         << OK at first loop call
<info> app: humidity:36.70                               << OK at first loop call
<info> app: pressure:97238.71
<info> app: raw_temperature:25.81
<info> app: raw_humidity:36.70
<info> app: gas:369129.00
<info> app: bsec_status:0                                   << 0 is ok ?
<info> app: *******************************
<info> app: systick:3231
<info> app: timestamp:2978000
<info> app: iaq:0.00                                        <<   0 ?
<info> app: iaq_accuracy:0                             <<   0 ?
<info> app: temperature:0.00                         <<  NOK
<info> app: humidity:100.00                           <<  NOK                         
<info> app: pressure:97237.11
<info> app: raw_temperature:25.87
<info> app: raw_humidity:36.72
<info> app: gas:424133.93
<info> app: bsec_status:0
<info> app: *******************************
<info> app: systick:6200
<info> app: timestamp:5947000
<info> app: iaq:0.00                                        <<   0 ?
<info> app: iaq_accuracy:0                             <<   0 ?
<info> app: temperature:0.00                         <<  NOK
<info> app: humidity:100.00                           <<  NOK
<info> app: pressure:97237.92
<info> app: raw_temperature:25.88
<info> app: raw_humidity:36.68
<info> app: gas:458585.78
<info> app: bsec_status:0
<info> app: *******************************
```
do you know this behavior?

Meroryan, 2017-10-20:
same with ..\Cortex_M4\libalgobsec.lib

BoschSensortec, 2017-10-23:
@Meroryan ,

Is your test code available on Github for review? Could you link it here?

Meroryan, 2017-10-23:
@BoschSensortec,
yes i uploaded it here:
https://github.com/Meroryan/sensor/tree/IAQ_Versuch

BoschSensortec, 2017-10-23:
@Meroryan ,

What are the values like after 5 minutes?

Meroryan, 2017-10-23:
pressure, raw_temperature, raw_humidity, gas and bsec_status shows plausible values.
iaq, iaq_accuracy and temperature stay at 0 and humidity at 100.00.
I have 1 board running since Friday. Noting changed until today.
if something with the timestamp would not be correct, then bsec_status would display it? 
why is it int64_t instead of unit64_t?

Meroryan, 2017-10-23:
could the output of  bsec_get_state() help?

BoschSensortec, 2017-10-23:
@Meroryan ,

Created a new issue for BSEC integration #7

closed 2017-11-21 by BoschSensortec