BoschSensortec/BME680_driver #5: /** BME680 I2C addresses */changed?

runkelz, 2017-10-12:
Regarding:
BME680_driver/bme680_defs.h (Version 3.51)
BME680_driver/bme680.h (Version 2.5)

I was not able to connect to the sensor with an ESP8266 unless I changed BME680_I2C_ADDR_PRIMARY to UINT8_C(0x77), which should be BME680_I2C_ADDR_SECONDARY. I followed the instructions in doc_bsec_step_by_step.md and it works exactly like described, except of the address, which I needed to change first. Version 3.51 of the driver still uses the same address.

BoschSensortec, 2017-10-12:
@runkelz ,

The I2C address on the BME680 is selected by the SDO line of the sensor. If SDO is low, the address is 0x76 and if high, 0x77. It seems that the SDO line in your setup is currently pulled high. If you would like to use the sensor via the primary interface, then you may set SDO to low.

runkelz, 2017-10-25:
Thank you, I wasn't aware of SDO-pins properties. I will try pulling it down and testing again.

closed 2017-11-21 by BoschSensortec