BoschSensortec/BME680_driver #6: Is gas output IAQ index?

forKnowYou, 2017-10-16:
There have some problems.
1.Can I use this formula to get a IAQ index?
rslt = bme680_get_regs(((uint8_t) (BME680_FIELD0_ADDR)), buff, (uint16_t) BME680_FIELD_LENGTH, dev);
data->gas = (float)((uint16_t) buff[13] << 2 | (uint16_t) buff[14] >> 6) / 1024 * 500;
Shoud I need another formula to get IAQ index?
2.Temp have a serious effect on pressure.
ex :    temp :24.38C  pressure :96053
         temp :25.36C  pressure :65904 (reduced 150)
It seems not normal.

Thanks for your reply.

Haini, 2017-10-16:
In my understanding you need to obtain the IAQ Gas Index by using the [BSEC Library](https://www.bosch-sensortec.com/bst/products/all_products/bsec) provided by Bosch. 
Otherwise you will only get the reading in Resistance of the Heater Plate in Ohm - which by itself already is somewhat an Index if you monitor it over multiple days and in different settings. 

I might want to Hijack this issue: Is there any contact information if you need the BSEC Library compiled for another CPU? It is fine that the Cortex ARMs are supported, even the ESP8266 but I would need a compiled version for the Raspberry Pi 1 Bs BCM2835 CPU. Are there any plans on releasing more binaries for different architectures? (especially with different compilation flags, e.g. Vector Floating Points (VFPs) via soft or hard settings etc.). 

Or can I open an own issue for this despite this being off topic for the API?

forKnowYou, 2017-10-17:
emmmm...
So I can't convert it(gas adc value) to IAQ Index?
Only can use statistical data for air quality Index?...
How about the effect between tempture and pressure?

BoschSensortec, 2017-10-17:
@forKnowYou ,

Regarding your query about the IAQ index, this is calculated by the BSEC software by interpreting the raw sensor data.

Regarding your query for the invalid pressure signal, could you send us a log of the communication when it is normal and when it fails? This will be helpful in finding the root cause of the issue. You can refer #4 for more information.

BoschSensortec, 2017-10-17:
Hi @Haini ,

If you would like it compiled for a specific platform, you can use our [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form) to get in touch with us for a special release and we can later add this to our standard releases. Also in the form, do mention any specific compilation options and some background as to why you think it is important to have it there.

hcs-svn, 2017-10-17:
Will you provide a way to to calculate the IAQ without using a precompiled lib or give information how the IAQ calculation is done?

BoschSensortec, 2017-10-17:
@hcs-svn ,

At this point of time the calculations for the Indoor Air Quality index are offered only as a pre-compiled library.

Haini, 2017-10-17:
@hcs-svn : I have logged the data over the past 16 hours and I think it is possible to make your own IAQ Index - under the premise that this is a hobby project. Just log some data for a certain scenario (e.g. 21 Degrees, 65% Humidity) and use it as a baseline for what is good. If you do that for some cases and write down how the air quality was during this time you will get a pretty good estimate. At least for a hobby project (with no sensor movement) that will suffice. 

@BoschSensortec : Thanks for the reply! I will do so.

forKnowYou, 2017-10-17:
I had modified it to adaptaion Arduino.
Can you test the example ?
https://github.com/DFRobot/DFRobot_BME680 

Thanks.

forKnowYou, 2017-10-18:
@BoschSensortec 
Did you find some problems?
Have you been faced the problem between tempture and pressure?

windkh, 2017-10-22:
The BME680 datasheet states

> Key parameters for gas sensor
 Response time (𝜏33−63%) < 1 s (for new sensors)
 Power consumption < 0.1 mA in ultra-low power mode
 Output data processing direct indoor air quality (IAQ) index output
<

Does this mean that the IAQ index can be read from one of the registers? 
If so could you supply a solution for how to get this IAQ index in Arduino based systems, please?

I would appreciate it very much if you could help me getting this to work to be able to integrate the sensor into a home automation solution.  (see also https://github.com/windkh/mysensors/tree/master/BME680Sensor)

nhandt3, 2017-11-06:
@BoschSensortec 
I filled out in your contact form for my BESC special release, how long that i will receive it?

windkh, 2017-11-11:
@BoschSensortec 
I filled out the contact form, too. But did not get a response yet.
Could you make the BSEC lib available for the Arduino?

wklenk, 2017-11-11:
@windkh I recently got the BSEC software library running in Arduino and a STM32F103 microcontroller board. It is a little bit tricky, but in the end it works. Just check my blog post here, maybe it will be helpful for you: https://wolfgangklenk.wordpress.com/2017/11/05/indoor-air-quality-iaq-measurement-with-bosch-bme680-and-stm32f103c8t6/

windkh, 2017-11-11:
@wklenk
Thanks, I already tried to get it running on an Arduino Uno, but the linker complains that the lib is incompatible... what is the difference between the uno and the stm32f103?

wklenk, 2017-11-11:
@windkh The Arduino Uno is a microcontroller board based on the ATmega328P. Unfortunately, these types of microcontrollers often have an 8-bit architecture, while "modern" microcontrollers have a 32-bit architecture. To be honest, I personally don't think that the BSEC software library will be backported from 32 bit to 8 bit architecture. 
I would recommend to take a look into modern 32-bit architecture supporting boards like ESP8266 or low-cost boards having a STM32 based microcontroller.

windkh, 2017-11-11:
@wklenk
I just wanted to do some experiments with the sensor, so the Uno was just at hand. Actually I just want to calculate the IAQ index, but unfortunately this algo is inside the BSEC lib. In my eyes there should be no need for a fast 32bit-CPU in a low power low and cost sensor scenario, that‘s the reason I want to have it running on the Arduino. Thanks for your effort!

Haini, 2017-11-12:
@windkh : Form looking at the source of the openly available preparation part it is highly unlikely there will be a backport to any 8 bit architecture. Using the sensor readings / resistance and the human brain to interpret these readings seems to do the job for a hobby project that is only used by me.  
@BoschSensortec : I have filled out my form too, yet no reaction.

windkh, 2017-11-12:
@Haini 
Thanks for your reply.
I’m already using a selfmade formula and the plotted curve comes close to what my MQ-135 sensor produces. Nevertheless I want to have it as accurate as possible... my inner engineer forces me into that direction ;)
What formula do you use?

windkh, 2017-11-12:
Here a little comparison between a MQ-135 and the BME680

[image]

radhoo, 2017-11-12:
Just to add to the discussion, here is the raw output comparison between
several setups.
82000094 and 8200009F are using an analogue VOC sensor, the MP503

The 820000B4, B5, B6, B7,B8 are using the BME680

These are all running in parallel in the same location. The values are read
every 5seconds, and averaged for an entire minute when the data is exported
and you can see it on the chart.

Completely different sensors, as you can see they pretty much follow the
same values (!!)

radhoo, 2017-11-12:
Just to add to the discussion, here is the raw output comparison between
several setups.
82000094 and 8200009F are using an analogue VOC sensor, the MP503

The 820000B4, B5, B6, B7,B8 are using the BME680

These are all running in parallel in the same location. The values are read
every 5seconds, and averaged for an entire minute when the data is exported
and you can see it on the chart.

Completely different sensors, as you can see they pretty much follow the
same values (!!)
[image]

windkh, 2017-11-12:
Well, there are two remarkable spikes in my plot: around 12:15 I was cooking. The BME680 is more sensitive here. While on 14:30 the situation is very different. Here I cleaned something using some kind of chemical liquid containing alcohol. Here the MQ is very sensitive but the BME is not. 20:30 I made a Pizza in my oven.

BoschSensortec, 2017-11-13:
@nhandt3 ,

Your request has been received and is currently being processed.

BoschSensortec, 2017-11-13:
@windkh ,

The BSEC library will not work on an Arduino Uno due to code size constraints and hence a release is not available at this time.

radhoo, 2017-11-13:
@windkh , I can confirm this in regards to alcohol.  I also observed that the BME680 is not very sensitive to methane, unlike other similar sensors (MiCS or the bigger ones like MQ).

BoschSensortec, 2017-11-13:
Hi @radhoo ,

How are you operating the sensor? Primarily, the heating temperature, heating duration and measurement frequency.

radhoo, 2017-11-13:
temperature to 360degrees
heater on for 100ms,
in forced mode

the measurement is repeated every 5 seconds. 

Are you considering higher temperature for breaking down methane?

BoschSensortec, 2017-11-13:
@radhoo ,

A temperature of 360 should be sufficient, although a longer duration of around 150ms should provide better results. Can you confirm?

radhoo, 2017-11-13:
Your suggestion makes sense. Although methane is not my target gas (I am looking for AQI as well) I will try that and report back.

windkh, 2017-11-13:
What is the maximum gas resistance?

BoschSensortec, 2017-11-14:
Hi @windkh ,

That is partially sensor dependent, but it should be between 12.5 to 13 Mohms.

nhandt3, 2017-12-09:
Hi @BoschSensortec 

I have plan using BME680 for my project, but it take too long time for waiting a BSEC special release, so please, re-confirm me when I receive it. I have to choose which air quality sensor is suitable as soon as posible.

BoschSensortec, 2017-12-11:
Hi @nhandt3,

Are you waiting for the RaspberryPi release?

nhandt3, 2017-12-11:
Hi @BoschSensortec 

I am waiting for ARM926EJ-S release. I can re-fill your contact form again if you need.

windkh, 2017-12-12:
@BoschSensortec 
Could you provide some information on how to calculate the IAQ index?
As there is no BSEC lib available for the arduino I would like to create an algorithm which would fit into the arduino memory. If you want to, we can meet in Reutlingen for a personal talk about this topic.

BoschSensortec, 2017-12-13:
Hi @windkh,

We will not disclose the exact nature of how the IAQ index can be calculated. In order to achieve the same performance as our BSEC algorithm, the code will not fit on Arduino boards based on the Atmel AVR series that have 32kB flash and 2kB of RAM.

We will be providing support for Arduino boards that have sufficient memory to run the algorithm like the ESP8266, ESP32, Cortex-M based MCUs. But for this to be released we need Arduino to support precompiled libs, a feature that  might be included in the next release of the Arduino IDE >1.8.5. Until then, we might release a version where one can modify the Arduino IDE inorder to use precompiled libs. This is currently under test and might be released in the coming days.

nhandt3, 2017-12-15:
Hi @BoschSensortec 

I re-filled your contact form again. Please, process it.

Thanks

BoschSensortec, 2017-12-15:
Hi @nhandt3,

We are working on it and are currently in the process of testing it. Please be patient. We will notify you when it will be available as a standard release.

juergs, 2018-01-19:
Meanwhile you released a 8Bit AVR library edition on Nov-17-2017. Towards which compiler/linker-version these libs are compiled?
regards
Jürgen

BoschSensortec, 2018-01-26:
Hi @nhandt3,

Kindly write to us at github@bosch-sensortec.com so we can send you a pre-release build of the requested library. 

Thank you.

juansebastian16, 2018-04-12:
Hi @BoschSensortec

I am using BME680 with a STM32F0 and I need to implement the IAQ calculation and to recognize other gases. How could I do that? because in the API  and datasheet I only see the gas_resistance calculation and no more.

juansebastian16, 2018-04-12:
Hi @Michal78S cool! How did you calculate it? Could you share your code or tell us something more about it?

abhinavbhargava, 2018-04-13:
I am using nrf52 arm cortex m4 and need to implement IAQ index on it could anybody suggest how ot go about it

BoschSensortec, 2018-04-16:
Hi @abhinavbhargava, 

Which IDE are you using?

abhinavbhargava, 2018-04-17:
I'm currently using Keil MicroVision Version 5.
Wtih NRF52 Microcontroller

BoschSensortec, 2018-04-17:
Hi @abhinavbhargava,

We do not have an out-of-the-box example for the NRF52 with Keil at the moment, but maybe the snippets [here](https://github.com/BoschSensortec/BSEC-Arduino-library/blob/master/src/bsec.cpp) in the wrapper can help you.

The binaries are available via our [website](https://www.bosch-sensortec.com/en/bst/products/all_products/bsec). The directory within the zip file is `algo\bin\Normal_version\armcc\Cortex_M4F\`

abhinavbhargava, 2018-04-17:
Hi

is there any implementation of this example in C and not C++(basically
without using classes).

WP-tim, 2018-04-21:
Just stumbled across this thread...
What is the status of retrieving the IAQ index on a Raspberry Pi Zero W ? (see Haini and nhandt3 above)

I am successfully retrieving the resistance value but need to convert  for submission to [wunderground](http://wiki.wunderground.com/index.php/PWS_-_Upload_Protocol).

WP-tim, 2018-04-22:
@alexh-name  Thanks.  Looking at your [python example](https://github.com/alexh-name/bme680_outputs), is there no way to import the library into python?

I am reluctant to write continuously to the SD Card; I've had corruption issue before.

alexh-name, 2018-04-22:
@WP-tim I think it would be better if you would open an issue in my repo, as I'm not affiliated with Bosch.

dude007dude007, 2018-05-07:
I am quite disappointed at the quality of documentation for the Bosch BME680, as the sensor does not give a VoC or AQI output but a changing resistance related to VoC.

_**The BME680 datasheet is missing:**_

-Sensitivity of the BME680 sensor to different VoCs as a function of plate temp and duration
-How to compensate for Gas Sensor Resistance Drift over Time or even acknowledge that it exists!
-Proper Burn-in procedure
-How to calculate an AQI with table
-How to convert the kOhm output into a usable and repeatable VoC ppm concentration value, despite 
      being uncalibrated 
-Disclaimer to User:  VoC  &  AQI will not work without a 32bit-only pre-compiled BSEC Library
-Suggestions to prevent premature failure or oven sensor fouling by bad settings, ie  (ie 1000ms @ 390C)
-An example of collecting a VoC reading and coverting kOhm to VoC ppm for a particular gas
-Calibration procedure for the BME680 using known VoC gas concentrations 

 
What if I sold a Temperature sensor that provided only a resistance which varied by how often you poll it in milliseconds, and then I don't ever supply a table with what resistance equals which temperature nor how to compensate for the resistance bias changing over time.  It would not be a temperature sensor anymore but a crappy thermister and I could not advertise nor sell it a Temperature sensor.  Especially, if I never give the table and equation to correct it to a Temperature reading due to the table being "proprietary".

I have been running a BME680 for three weeks continuously at 1 second polling  (oven 320C @ 150ms) and have seen the resistance change from 80kOhm an hour after burn-in to the current value of 1.350 Mohm, which continues to slowly climb.  (See the attached graph)   There is no explanation for this increase in the datasheet nor information on how to correct it.  This makes the BME680 VoC sensor completely unusable for repeatable measurements.  If I'm in error somewhere, feel free to educate me.  Thanks.
[image]

ourairquality, 2018-05-07:
I share your frustration, and echo it. What I am trying is to do is record the raw values supplied rather than applying any processing, not even applying the calibration. The calibration data is also logged on startup. This allows post-correction for the ADC discontinuities, re-running the calibration calculations on the data, and experimenting with the analysis. Also recording multiple temperature/time results, just two at present. The timing is important, and a rtos is used and it goes to lengths to try to time the measurements at consistent periods, and the actual measurement time is also logged to a precision of of around 10msec which seems adequate to validate the timing. Still trying to understand the data and what use it is, but here is just one example where it seems to help - a significant but short pollution even where the PM2.5 climbed to very poor levels and the BEM680 conductance followed that. At other times it seems to vary for no apparent reason, but their might be a reason? The sensor does seem to take time to settle at start up, even a brief outage takes time to recover, and a longer outage seems to take almost a day to recover from. Have not explored burn-in methods yet to see if anything could speed recovery. Longer term drift also looks likely, not certain. The sensor does respond at least in some cases, and it responds relatively quickly, it has given much clearer signals than some electrochemical and diffusive optical sensors side-by-side. There are growing citizen science air quality networks, they seem to be setting on the bme280, and wouldn't it be great to be able to simply upgrade to the bme680 and add some VOC information! I've purchased more for testing, but it is disappointing that it seems we need to explore the merits of the bme680 on our own, and the limited disclosure from Bosch. The IAQ library is of no value to me, where post analysis and transparency are critical, shall have to work something out on my own. It might be that the concept of placing a heated VOC sensor so close to the temperature and humidity sensor is not going to be proven a good design, who knows, but it's an easy way to start playing with a VOC sensor.
[image]

ourairquality, 2018-05-08:
Fwiw here is another event showing a plume of pollution producing some peaks in PM that the VOC sensor responds to nicely, but in this case on top of a larger fluctuation in the sensor conductance.

This might be a case of lacking selectivity, but not sure that is a problem with the sensor. There was no clear correlation to humidity or temperature, they might have been factors, but it was not obvious how to separate them in this one example.

It still provides some support for other air quality measurements, and as noted before it responds quickly enough to see such quick air quality changes. A NDIR CO2 sensor did not respond as clearly, and an electro chemical sensor had no clear response over its noise level.

One point I find interesting is that the VOC sensor appears to respond in both directions. Would that be to reducing and oxidising gasses moving it in different directions? Or is there a balance in standard air that is disturbed? We don't appear to even have any documentation on what to expect of the sensor to various gasses? Another possibility is that exposure contaminated the sensor pushing it in a different direction? Might the rate of response be different in either direction?

The question of reproducibility might be in two parts: 1. absolute VOC level. 2. magnitude of variation.

I wonder if some periodic probing at a range of temperatures could help calibrate the later where the air composition is also naturally varying?
[image]

CarlosGS, 2018-06-29:
Hi, first of all greetings to Bosch for the neat sensors and adequate communication with us.
I'd like to ask for one particular thing already commented above:
* _Suggestions to prevent premature failure or oven sensor fouling by bad settings, ie (ie 1000ms @ 390C)_

This is the most crucial to me, as the sensors are expensive and would like to prevent damage to them. In fact, if we damage our sensors unknowingly we could get bad results and give you unfair feedback!

Could you briefly state some examples of what settings should be avoided? Ideally it would be a graph of temperatures/durations that shows harmful combinations and shows the region of "OK" values. _Let's keep it simple by assuming "normal" air without strong odors near the sensor._

Also I would appreciate your input on the effect that bad settings or element burning could have on the output. Mistakenly burnt sensors change their range? sensitivity? (anything apparent that we can look for?)

Thank you, and kind regards!

dsyleixa, 2018-08-03:
hi,
I still do not understand: 
is there a working IAQ lib for esp8266 or  SAMD21 (M0 Zero) or SAMDX8 (Due)   and/or perhaps even  the Mega2560 (i2c) working with the Arduino IDE?
Where can I see the documentation?
What meaning do the IAQ values actually have?

BoschSensortec, 2018-08-07:
Hi @dsyleixa,

You can refer [this](https://github.com/BoschSensortec/BSEC-Arduino-library ) dedicated repository for your request.

dsyleixa, 2018-11-12:
thx, but sry, I do not understand yet ... :
- in which respect is the BME680 related to BSEC?
- what does Air Quality actually mean in detail?
and above all: 
how can I poll the specific ppm values for 
- methane, 
- ethane, 
- propane, 
- butane,
- diethylether, 
- acetone,
- CO2, 
- CO, 
- NOx, 
- and perhaps even more specific gas ppm values?

Kormykake, 2019-01-15:
One vote from me to VFP support for Cortex M4F.  I'am using nRF52 and they are using (-mfloat-abi=hard) option in their Makefile that seems to cause "VFP" compilation error. I tried to switch the flag to soft or softfp to remove the error (as proposed in many web discussion to remove the "VFP" error message)...Didn't help maybe because Nordicsemi have likely compiled their crypto-libraries with the hard flag (I quess - I'am not the compiler expert).