BoschSensortec/BME680_driver #7: BSEC integration with NRF52

BoschSensortec, 2017-10-23:
Take over from #4

BoschSensortec, 2017-10-23:
Answering previous questions, int64_t is used to easily detect an overflow (time < 0).  bsec_get_state and bsec_set_state are used for saving the operation state in flash and resuming it after a longer period of time or a restart. 
The calibration procedure takes around 5 minutes to complete and for the first run, IAQ and IAQ accuracy will not be provided as an output. After 5 minutes, once IAQ accuracy != 0, you can use get_state and save it in flash. In addition you can save it around twice a day or before a power down of the device.

The strange readings for SENSOR_HEAT_COMPENSATED_TEMPERATURE and SENSOR_HEAT_COMPENSATED_HUMIDITY is unusual. Are you setting the value of temperature_offset? If so, what is the value? This is used to indicate to the algorithm what is the effect of heat sources of components on the board (power supply, mcu, etc) towards the temperature sensor.

Meroryan, 2017-10-23:
> Are you setting the value of temperature_offset? If so, what is the value?
i call:
bsec_iot_init(BSEC_SAMPLE_RATE_LP, **0.0f**, bus_write, bus_read, sleep);

Meroryan, 2017-10-24:
I think I found the problem:
befor
```
int64_t get_timestamp_us(void)
{ 
    static int64_t system_current_time = 0; 
    system_current_time	= sys_time; 
    return system_current_time; 
} 
```
after
```
int64_t get_timestamp_us(void)
{
    int64_t system_current_time = 0;
    system_current_time	= (int64_t)sys_time * 1000;
    return system_current_time;
}
```
the 2nd seems working.
I mistook ms and us.