BoschSensortec/BME680_driver #8: API produces different pressure values between rev 3.5.1 and rev 2.2.0

skaringa, 2017-10-31:
When I use the actual version 3.5.1 of the api, then I get wrong pressure values from calc_pressure.
For example, in my actual condition I get values of 1022 hPa, which is much to high. I would expect values of 1014 hPa.
The strange thing is, if I use the BSEC software with the bundled driver 2.2.0, then I get much more meaningful values of 1016 hPa.
I use the same sensor under the same conditions both times!
The difference seems to be caused by the function calc_pressure() of 3.5.1 compared with bme680_compensate_pressure_double() that is used by BSEC/driver 2.2.0.
Could you explain what is happening here?

BoschSensortec, 2017-11-02:
@skaringa ,

It seems like there seem to be differences when compiling the code with different compilers. There is an upcoming release which uses the same calculation code as in v2.x

BoschSensortec, 2017-11-21:
@skaringa ,

Reverted compensation equations to v2.x in v3.5.3 onwards.

closed 2017-11-21 by BoschSensortec