BoschSensortec/BME680_driver #11: Can't read data with SPI interface

forKnowYou, 2017-11-29:
Hi, we had read correct IAQ on ESP8266 successful with I2C interface, thanks for your package [BSEC_1341_Generic_Release_20170721.zip.](https://ae-bst.resource.bosch.com/media/_tech/media/bsec/BSEC_1341_Generic_Release_20170721.zip)
but there has some problem with SPI interface.

We can't read tempture, humidity and pressure but iaq. eg print.
iaq :51.99
temp :-0.05
humi :0.00
pre :0.00

here is my arduino code on ESP8266
`
static int8_t bme680_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *data, uint8_t len)
{
  SPI.begin();
  digitalWrite(bme680_cs, 0);
  SPI.transfer(reg_addr | 0x80);
  while(len --) {
    *data = SPI.transfer(0x00);
    data ++;
  }
  digitalWrite(bme680_cs, 1);
  return 0;
}

static int8_t bme680_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *data, uint8_t len)
{
  SPI.begin();
  digitalWrite(bme680_cs, 0);
  SPI.transfer(reg_addr);
  while(len --) {
    SPI.transfer(*data);
    data ++;
  }
  digitalWrite(bme680_cs, 1);
  return 0;
}

static void bme680_delay_ms(uint32_t period)
{
  delay(period);
}

static int64_t bme680_getus(void)
{
  return (millis() * 1000);
}

void bme680_outputReady(int64_t timestamp, float iaq, uint8_t iaq_accuracy, float temperature,
                        float humidity, float pressure, float raw_temperature, float raw_humidity,
                        float gas, bsec_library_return_t bsec_status)
{
  Serial.print("time :");
  Serial.println((uint32_t)timestamp);
  Serial.print("iaq :");
  Serial.println(iaq);
  Serial.print("temp :");
  Serial.println(temperature);
  Serial.print("humi :");
  Serial.println(humidity);
  Serial.print("pre :");
  Serial.println(pressure);
}

void setup(void)
{
  Serial.begin(115200);
  delay(200);
  bsec_iot_init(0.33333f, 0, bme680_spi_write, bme680_spi_read, bme680_delay_ms, 0x77, 
BME680_SPI_INTERFACE);  //0x77 dev_id, BME680_SPI_INTERFACE interface
  bsec_iot_loop(bme680_delay_ms, bme680_getus, bme680_outputReady);
}

void loop(void) {}
`
Change in bsec_iot_init;
`
void bsec_iot_init(float sample_rate, float temperature_offset, sensor_write bus_write, sensor_read bus_read, 
                   sleep_fct sleep, uint8_t addr, uint8_t interface)
{
    enum bme680_return_type bme680_status = BME680_COMM_RES_OK;
    bsec_library_return_t bsec_status = BSEC_OK;
    
    /* Fixed I2C configuration */
    //bme680_g.dev_addr = BME680_I2C_ADDR_PRIMARY;
    bme680_g.dev_addr = addr;
    bme680_g.interface = interface;
    /* User configurable I2C configuration */
    bme680_g.bme680_bus_write = bus_write;
    bme680_g.bme680_bus_read = bus_read;
    bme680_g.delay_msec = sleep;
    
    /* Initialize BME680 API */
    bme680_status = bme680_init(&bme680_g);
    
    /* Initialize BSEC library */
    bsec_status = bsec_init();
    
    /* Set temperature offset */
    bme680_temperature_offset_g = temperature_offset;
    
    /* Call to the function which sets the library with subscription information */
    bsec_status = bme680_bsec_update_subscription(sample_rate);
}
`
anything else no change.
can you give us some help?

BoschSensortec, 2017-11-29:
@forKnowYou ,

Can you check with the following snippet?

```c
SPISettings settg_bme(1000000, MSBFIRST, SPI_MODE0); // 1MHz clock speed. Upto 10MHz is supported by the sensor

static int8_t bme680_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *data, uint8_t len)
{
  SPI.beginTransaction(settg_bme);
  digitalWrite(bme680_cs, LOW); // Could be dev_id instead of bme680_cs
  SPI.transfer(reg_addr | 0x80);
  for(uint16_t i = 0; i < len; i++) {
    data[i] = SPI.transfer(0x00); // Don't modify the value of the pointer
  }
  digitalWrite(bme680_cs, HIGH); // Could be dev_id instead of bme680_cs
  SPI.endTransaction();
  
  return 0;
}

static int8_t bme680_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *data, uint8_t len)
{
  SPI.beginTransaction(settg_bme);
  digitalWrite(bme680_cs, LOW); // Could be dev_id instead of bme680_cs
  SPI.transfer(reg_addr);
  for(uint16_t i = 0; i < len; i++) {
    SPI.transfer(data[i]); // Don't modify the value of the pointer
  }
  digitalWrite(bme680_cs, HIGH); // Could be dev_id instead of bme680_cs
  SPI.endTransaction();
  
  return 0;
}

static void bme680_delay_ms(uint32_t period)
{
  delay(period);
}

static int64_t bme680_getus(void)
{
  return (millis() * 1000);
}

void bme680_outputReady(int64_t timestamp, float iaq, uint8_t iaq_accuracy, float temperature,
                        float humidity, float pressure, float raw_temperature, float raw_humidity,
                        float gas, bsec_library_return_t bsec_status)
{
  Serial.print("time :");
  Serial.println((uint32_t)timestamp);
  Serial.print("iaq :");
  Serial.println(iaq);
  Serial.print("temp :");
  Serial.println(temperature);
  Serial.print("humi :");
  Serial.println(humidity);
  Serial.print("pre :");
  Serial.println(pressure);
}

void setup(void)
{
  Serial.begin(115200);
  SPI.begin();
  pinMode(bme680_cs, OUTPUT);
  delay(200);
  bsec_iot_init(0.33333f, 0, bme680_spi_write, bme680_spi_read, bme680_delay_ms, 0, BME680_SPI_INTERFACE);  //dev_id or dev_addr should contain the chip select, bme680_cs. 0 in this example
  bsec_iot_loop(bme680_delay_ms, bme680_getus, bme680_outputReady);
}

void loop(void) {}
```

forKnowYou, 2017-11-29:
@BoschSensortec 
faild.....
you successed, do you?...

BoschSensortec, 2017-11-29:
@forKnowYou ,

Yes it works, can you send us more details about your hardware setup?

forKnowYou, 2017-11-30:
@BoschSensortec 
[image]
this is the hardware.

BoschSensortec, 2017-11-30:
@forKnowYou ,

Do you have a logic trace or a oscilloscope screenshot of the communication? It is difficult to guarantee the operating clock frequency of the SPI due to all the pull up resistors.

forKnowYou, 2017-12-01:
@BoschSensortec 
I had print SPI read/write process. I think it can read data and not analysis correctly. my lib version is V2.2.0

here is the print.
'''
SPI read :73 10 

SPI write :73 10 

SPI read :70 0 10 1 10 54 0 

SPI write :70 0 71 10 72 1 73 10 74 54 75 0 

SPI read :73 10 

SPI write :73 10 

SPI read :74 54 

SPI write :5A CE 5B 0 5C 0 5D 0 5E 0 5F 0 60 0 61 0 62 0 63 0 64 71 65 0 66 0 67 0 68 0 69 0 6A 0 6B 0 6C 0 6D 0 6E 0 

SPI read :74 54 

SPI write :74 55 

SPI read :74 54 

SPI read :73 10 

SPI write :73 10 

SPI read :1D 80 

SPI read :1D 0 0 57 25 E0 71 E8 0 56 33 80 0 0 C4 B7 

time :3658000
iaq :0.00
temp :-0.02
humi :0.00
pre :0.00
'''
first data is reg_addr, flow is read/write data.

BoschSensortec, 2017-12-01:
@forKnowYou ,

Can you update to the latest version of the sensor API?

BoschSensortec, 2017-12-01:
@forKnowYou ,

One of the newer versions of the API v3.5.3 is already integrated in the latest version of the bsec IoT example which can be downloaded from our website.

forKnowYou, 2017-12-01:
@BoschSensortec 
downloaded from [here.](https://www.bosch-sensortec.com/en/bst/products/all_products/bme680)
there is some bug in bsec_iot_example.ino
and if update this code, it makes my ESP8266 reset again and again...

'''

int8_t bus_write(uint8_t dev_addr, uint8_t reg_addr, uint8_t *reg_data_ptr, uint16_t data_len)
{
    Wire.beginTransmission(dev_addr);
    Wire.write(reg_addr);    /* Set register address to start writing to */
 
    /* Write the data */
    for (int index = 0; index < data_len; index++) {
        Wire.write(reg_data_ptr[index]);
    }
 
    return (int8_t)Wire.endTransmission();
}

int8_t bus_read(uint8_t dev_addr, uint8_t reg_addr, uint8_t *reg_data_ptr, uint16_t data_len)
{
    int8_t comResult = 0;
    Wire.beginTransmission(dev_addr);
    Wire.write(reg_addr);                    /* Set register address to start reading from */
    comResult = Wire.endTransmission();
 
    delayMicroseconds(150);                 /* Precautionary response delay */
    Wire.requestFrom(dev_addr, data_len);    /* Request data */
 
    int index = 0;
    while (Wire.available())  /* The slave device may send less than requested (burst read) */
    {
        reg_data_ptr[index] = Wire.read();
        index++;
    }
 
    return comResult;
}

void sleep(uint32_t t_ms)
{
    delay(t_ms);
}

int64_t get_timestamp_us()
{
    return (int64_t) millis() * 1000;
}

void output_ready(int64_t timestamp, float iaq, uint8_t iaq_accuracy, float temperature, float humidity,
     float pressure, float raw_temperature, float raw_humidity, float gas, bsec_library_return_t bsec_status)
{
      Serial.print("[");
      Serial.print(timestamp/1e6);
      Serial.print("] T: ");
      Serial.print(temperature);
      Serial.print("| rH: ");
      Serial.print(humidity);
      Serial.print("| IAQ: ");
      Serial.print(iaq);
      Serial.print(" (");
      Serial.print(iaq_accuracy);
      Serial.println(")");
}

void saveState(const uint8_t *state_buffer, uint32_t length) {}

uint32_t loadState(uint8_t *state_buffer, uint32_t n_buffer) {}

uint32_t loadConfig(uint8_t *state_buffer, uint32_t n_buffer) {}

void setup()
{
    /* Init I2C and serial communication */
    Wire.begin();
    Serial.begin(115200);
  
    /* Call to the function which initializes the BSEC library 
     * Switch on low-power mode and provide no temperature offset */
    bsec_iot_init(BSEC_SAMPLE_RATE_LP, 0.0f, bus_write, bus_read, sleep, loadState, loadConfig);
    
    /* Call to endless loop function which reads and processes data based on sensor settings */
    bsec_iot_loop(sleep, get_timestamp_us, output_ready, saveState, 0);
}

void loop()
{
}
'''

forKnowYou, 2017-12-01:
@BoschSensortec 
thanks for your help, it can be work correct now.

closed 2017-12-01 by forKnowYou