BoschSensortec/BME680_driver #13: How can I use libalgobsec.a file on avr board(arduino) and esp32 board(arduino)?

forKnowYou, 2017-12-05:


BoschSensortec, 2017-12-05:
Hi @forKnowYou,

The current release will not fit on an avr board like the Uno but only on a MEGA2560. Currently there is no release for the ESP32 either. We will add these in upcoming releases.

forKnowYou, 2017-12-05:
OK, thanks.

forKnowYou, 2017-12-05:
How can use libalgobsec.a on mega 2560?
I can't find any way to config it.

BoschSensortec, 2017-12-05:
@forKnowYou,

Do you mean in the Arduino IDE? This is quite complicated as the Arduino IDE currently doesn't support precompiled libs. This will probably be supported in on of their next releases. Once they do, we will also release a BSEC library for use in the Arduino IDE with examples.

c-t-d, 2018-01-25:
Hi BoschSensortec,
is it correct that there will not be a BSEC library for the arduino uno/nano/etc. in the near future?
btw. one can link static libraries with the Arduino IDE by placing the library into the appropriate folder and adding -lalgobsec to the platform.txt similar to the steps for ESP. However, i don't know whether the build binary actually works on the arduino.
Thanks for your help!

BoschSensortec, 2018-01-25:
Hi @c-t-d,

We provide BSEC binaries for AVR microcontrollers, both 8bit and 32bit. The issue is that a fully functioning application, as of today, will not fit in the limited memory spaces of the uno/nano etc. On the other hand it is possible to create a working application with a board like the MEGA 2560.

Regarding the use of binaries in the Arduino IDE, one has to 'hack' the platform.txt file and copy the binary to a defined location. You can observe a similar mod mentioned in the examples for the ESP8266. That being said, a future version of the Arduino IDE will support easier integration of precompiled libraries, almost the same as with existing libraries, but this will only be supported on a newer version of the [Arduino builder](https://github.com/arduino/arduino-builder). It seems like the Arduino Beta for 1.9.x should have this updated builder integrated in the coming weeks. We will release an easy-to-use Arduino library shortly after this feature is available.

c-t-d, 2018-01-25:
Hi @BoschSensortec,
ok. Many thanks for the clarification!

michapr, 2018-02-10:
Hello,
have you any documentation / steps how to integrate the BSEC for the ATMEGA and other AVR in Arduino IDE? For ESP it is shown - and working. I see that you have precompiled libs for AVR8_megaAVR, AVR8_XMEGA, AVR32.
BTW: will there be in future a smaller lib for ATMEGA 328 etc? ;)

Thanks!
Michael

BoschSensortec, 2018-02-12:
Hi,

In order to use the example with AVR, you need to instruct the compiler where to find the precompiled library. In the current release, using the existing examples, you need to modify the platform.txt file usually found at ```{yourArduinoDirectory}/hardware/arduino/avr``` as below. Line 67 and 68 [(see link)](https://github.com/arduino/Arduino/blob/3d3bb385ad7575596b07ca20bf55d50a0166a215/hardware/arduino/avr/platform.txt#L67) should be replaced by
```
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm -L{runtime.platform.path}/libraries/bsec/{build.mcu} -lalgobsec
```

Then in ```{yourArduinoDirectory}/hardware/arduino/libraries```, create a directory ```bsec``` and 2 sub-directories ```atmega1280``` and ```atmega2560```

In both the sub-directories, add the precompiled lib ```libalgobsec``` from the path ```algo\bin\avr\AVR8_megaAVR```  in the zip folder that you can download from our [website](https://www.bosch-sensortec.com/en/bst/products/all_products/bsec)

Surroot, 2018-03-17:
Hi,

are there any news on this?
I've tried to build my project for an AtMega 328P with the precompiled library and failed of course, then I found this thread.
I've tried to use the IAQ functionality of the [DFRobot_BME680](https://github.com/DFRobot/DFRobot_BME680) implementation of your sensor, but it fails to build for this MCU. I get following error:
`avr/bin/ld: skipping incompatible lib/libalgobsec.a when searching for -lalgobsec`

But I don't understand why I can't use the precompiled library. Is the IAQ functionality so huge? (Sorry but I have no idea how such a precompiled library works... Is it necessary to load the full library onto the MCU or is it like with normal libraries and only the used functions are compiled and built into the hex file?)

Thanks!
Philipp

BoschSensortec, 2018-03-19:
Hi @Surroot,

We will be creating a new repository for BSEC integration and tutorials to integrate the same in the coming weeks.

As mentioned earlier, the atmega328p cannot fit the required functions of the library to get an IAQ estimate. The error you are receiving is due to the fact that you are linking the wrong static library to your project.

Surroot, 2018-03-20:
Hi,

> We will be creating a new repository for BSEC integration and tutorials to integrate the same in the coming weeks.

Thanks! I look forward to it.

> As mentioned earlier, the atmega328p cannot fit the required functions of the library to get an IAQ estimate.

Well, I read that, but I couldn't believe it... I can't believe its that huge, but well I don't know what is involved in this calculation. So I have no chance but to use a different MCU when I want to use the IAQ calculation...

> The error you are receiving is due to the fact that you are linking the wrong static library to your project. 

Hmm. No, I definitely used the right "libalgobsec.a". Another user of the platformio forum also tried to use it and managed to compile it for 2560 but not for 328P. Please have a look at the [forum](https://community.platformio.org/t/using-a-precompiled-library-a/3405/9).
But it does not matter anymore, as the library can't be used for 328P.

Thanks for your answer!

Best regards,
Philipp

jttaylor86, 2018-03-29:
Hello,

I am trying to integrate the bsec algorithm with no success currently. I am using an ESP32.

Which .ld file do I need to change? 

I have added -lalgobsec to the platform.txt file
I have added the libalgobsec.a to tools/sdk/lib

Thanks,

Justin

juampe, 2018-04-09:
A guide to give more clues about this IAQ >400
https://wolfgangklenk.wordpress.com/2017/11/05/indoor-air-quality-iaq-measurement-with-bosch-bme680-and-stm32f103c8t6/

BoschSensortec, 2018-04-10:
Hi @jttaylor86,

You can check the integration guide in our library dedicated to using the BSEC library in Arduino.

juampe, 2018-04-13:
Finally I got, with ESP8266 NodeMCU
Development environment Raspberri Pi Raspbian 8 and arduino-1.8.5 (any Debian, or standard Linux must be valid)

**Base software**. Maybe firsly apt-get -y install arduino to install JRE and dependencies. But I try to do a bug free, step by step HowTo ;)
```
cd ~/
wget https://downloads.arduino.cc/arduino-1.8.5-linuxarm.tar.xz
tar -xvJf https://downloads.arduino.cc/arduino-1.8.5-linuxarm.tar.xz
~/arduino-1.8.5/arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs 2>&1
~/arduino-1.8.5/arduino --install-boards esp8266:esp8266 2>&1
```

```
mkdir ~/arduino
cd ~/arduino
wget https://ae-bst.resource.bosch.com/media/_tech/media/bsec/BSEC_1.4.6.0_Generic_Release_20180409.zip
unzip BSEC_1.4.6.0_Generic_Release_20180409.zip
mv BSEC_1.4.6.0_Generic_Release_20180409 BSEC
```

```
cp ~/arduino/BSEC/algo/bin/Normal_version/esp8266/libalgobsec.a ~/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/lib/
```

**Fix linker script**
~/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/ld/eagle.app.v6.common.ld
Add the llibalgobsec.a line to include it to the ROM

```
    *libc.a:(.literal .text .literal.* .text.*)
    *libm.a:(.literal .text .literal.* .text.*)
    *libgcc.a:_umoddi3.o(.literal .text)
```
to
```
    *libc.a:(.literal .text .literal.* .text.*)
    *libm.a:(.literal .text .literal.* .text.*)
    *libalgobsec.a:(.literal .text .literal.* .text.*)
    *libgcc.a:_umoddi3.o(.literal .text)
```

**Fix arduino esp8266 build**
~/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/platform.txt

add the second and third line
```
build.led=
build.flash_ld=eagle.flash.4m.ld
build.flash_size=4M
```
And add -lalgobsec to the compiler.c.elf.libs after -lm
```
compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm  -lc -lgcc
```
change to
```
compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lalgobsec -lc -lgcc
```

**Fix OTA script**
```
chmod 755 ~/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/espota.py
```

**Fix esptool**, Only for Raspberry pi 1 armel architecture  esptool (no problem with newer RPi3 armhf, skip this step )
```
cd ~/.arduino15/packages/esp8266/tools/
git clone https://github.com/igrr/esptool-ck.git
make
cp esptool ~/.arduino15/packages/esp8266/tools/esptool/0.4.13/esptool
[... 136 lines cut ...]
        -warnings=none \
        -prefs=build.warn_data_percentage=75 \
        -verbose "$(SKETCH)"

flash:
        $(ESPTOOL) -v -cd nodemcu -cb 230400 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

ota:
        $(ESPOTA) -i $(OTA_IP) -I 192.168.1.10 -a $(OTA_PASSWD) -f $(TARGET_DIR)/$(SKETCH).bin

clean:
        rm -rf $(TARGET_DIR)

monitor:
        screen $(MONITOR_PORT) 115200

```

**To finish**
```
make
make flash
```

#Enjoy