BoschSensortec/BME680_driver #14: BSEC 1.4.5 and raspbian

axelbusch, 2017-12-14:
Hi,

I try to run the BME680 on a raspberry pi using the driver API v3.5.5 and the libalgobsec for c. I implemented the following three header files as follows:
```
void user_delay_ms(uint32_t period)
{
    usleep(period*1000);
}

uint8_t user_i2c_Begin(uint8_t dev_id){
    int fd;
    char *fileName = "/dev/i2c-1";
        if ((fd = open(fileName, O_RDWR)) < 0)
                exit(1);
        if (ioctl(fd, I2C_SLAVE, dev_id) < 0) {
                close(fd);
                exit(1);
        }
    return fd;
}

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len){
    int8_t rslt = 0;
    uint8_t fd = user_i2c_Begin(dev_id);

    if(i2c_smbus_read_i2c_block_data(fd, reg_addr, len, reg_data) < 0) {
        rslt = -1;
    }
    close(fd);

    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len){  
    int8_t rslt = 0;
    uint8_t fd = user_i2c_Begin(dev_id);

    if (i2c_smbus_write_block_data(fd, reg_addr, len, reg_data) < 0) {
                rslt = -1;
        }
    close(fd);

    return rslt;
}
```

Unfortunately I always get only one measurement (what is wrong actually):

> T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms
T: 33.59 degC, P: 676.72 hPa, H 64.74 %rH , G: 12976914 ohms

What am I doing wrong here?

axelbusch, 2017-12-14:
I added now 
```
void handle_bme680_error(int8_t _rslt)
{
    if (_rslt < BME680_OK) {
        printf("BME680 Error code: %d\n", _rslt);
        for(;;); /* Trap if error */
    } else if (_rslt > BME680_OK) {
        printf("BME680 Warning code: %d\n", _rslt);
    }
}
```
This results in
> BME680 Warning code: 2

Further, I added after reading:
```
    printf("RD: %d, %d", reg_addr, len);
    for(uint16_t idx = 0; idx < len; idx++)
        printf(" 0x%x", reg_data[idx]);
    printf("\r\n");
```

and before writing:
```
printf("WR: %d, %d", reg_addr, len);
    for(uint16_t idx = 0; idx < len; idx++)
        printf(" 0x%x", reg_data[idx]);
    printf("\r\n");
```

and got

> WR: 224, 1 0xb6
RD: 208, 1 0x61
RD: 137, 25 0x80 0x55 0x67 0x3 0x2f 0xfd 0x8f 0xd7 0xd6 0x58 0xff 0xc7 0x1f 0x90 0xff 0x3a 0x1e 0x0 0x0 0x8 0xee 0xfc 0xf8 0x1e 0x3
RD: 225, 16 0x40 0x19 0x2b 0x0 0x2d 0x14 0x78 0x9c 0x0 0x66 0x12 0xa7 0xea 0x12 0x71 0x0
RD: 2, 1 0x16
RD: 0, 1 0x32
RD: 4, 1 0xf3
WR: 90, 3 0x6c 0x64 0x65
RD: 116, 1 0x0
RD: 117, 1 0x7
RD: 116, 1 0x0
RD: 114, 1 0x2
RD: 113, 1 0x10
WR: 117, 7 0xb 0x74 0x8c 0x72 0x2 0x71 0x10
RD: 116, 1 0x0
WR: 116, 1 0x1
RD: 29, 15 0x80 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
T: 33.59 degC, P: 676.72 hPa, H 64.73 %rH , G: 12976914 ohms
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20
RD: 29, 15 0x0 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0xa9 0x80 0x0 0x0 0x0 0x20

BoschSensortec, 2017-12-15:
Hi @axelbusch ,

It seems like you are not triggering a measurement. Can you attach the snippet where you are configuring and reading the sensor data?

axelbusch, 2017-12-15:
Hi @BoschSensortec ,

thank you for the quick reply. I use the code snippet from the bme680 driver example:
```
#include "bsec_integration.h"
#include <stdio.h>
#include <stdlib.h>
#include "bme680.h"
#include <unistd.h>
#include <sys/ioctl.h>
#include <linux/i2c-dev.h> 
#include <fcntl.h>
//templates omitted here
int main()
{
    struct bme680_dev gas_sensor;

	gas_sensor.dev_id = BME680_I2C_ADDR_PRIMARY;
	gas_sensor.intf = BME680_I2C_INTF;
	gas_sensor.read = user_i2c_read;
	gas_sensor.write = user_i2c_write;
	gas_sensor.delay_ms = user_delay_ms;

	int8_t rslt = BME680_OK;
	rslt = bme680_init(&gas_sensor);
     	uint8_t set_required_settings;

	/* Set the temperature, pressure and humidity settings */
	gas_sensor.tph_sett.os_hum = BME680_OS_2X;
	gas_sensor.tph_sett.os_pres = BME680_OS_4X;
	gas_sensor.tph_sett.os_temp = BME680_OS_8X;
	gas_sensor.tph_sett.filter = BME680_FILTER_SIZE_3;

	/* Set the remaining gas sensor settings and link the heating profile */
	gas_sensor.gas_sett.run_gas = BME680_ENABLE_GAS_MEAS;
	/* Create a ramp heat waveform in 3 steps */
	gas_sensor.gas_sett.heatr_temp = 320; /* degree Celsius */
	gas_sensor.gas_sett.heatr_dur = 150; /* milliseconds */

	/* Select the power mode */
	/* Must be set before writing the sensor configuration */
	gas_sensor.power_mode = BME680_FORCED_MODE; 

	/* Set the required sensor settings needed */
	set_required_settings = BME680_OST_SEL | BME680_OSP_SEL | BME680_OSH_SEL | BME680_FILTER_SEL 
		| BME680_GAS_SENSOR_SEL;
		
	/* Set the desired sensor configuration */
	rslt = bme680_set_sensor_settings(set_required_settings,&gas_sensor);

	/* Set the power mode */
	rslt = bme680_set_sensor_mode(&gas_sensor);

	/* Get the total measurement duration so as to sleep or wait till the
	 * measurement is complete */
	uint16_t meas_period;
	bme680_get_profile_dur(&meas_period, &gas_sensor);
	user_delay_ms(meas_period); /* Delay till the measurement is ready */

	struct bme680_field_data data;
	
	while(1) 
	{
		rslt = bme680_get_sensor_data(&data, &gas_sensor);
		handle_bme680_error(rslt);
		printf("T: %.2f degC, P: %.2f hPa, H %.2f %%rH ", data.temperature / 100.0f,
			data.pressure / 100.0f, data.humidity / 1000.0f );
		/* Avoid using measurements from an unstable heating setup */
		if(data.status & BME680_GASM_VALID_MSK)
			printf(", G: %d ohms", data.gas_resistance);
		printf("\r\n");
	}
    
    return 0;
}
```

BoschSensortec, 2017-12-15:
Hi @axelbusch ,

Some of the code should be moved inside the while loop,

```c
while(1) 
	{
                /* Set the power mode */
	        rslt = bme680_set_sensor_mode(&gas_sensor);

	        /* Get the total measurement duration so as to sleep or wait till the
	         * measurement is complete */
	        uint16_t meas_period;
	        bme680_get_profile_dur(&meas_period, &gas_sensor);
	        user_delay_ms(meas_period); /* Delay till the measurement is ready */

		rslt = bme680_get_sensor_data(&data, &gas_sensor);
		handle_bme680_error(rslt);
		printf("T: %.2f degC, P: %.2f hPa, H %.2f %%rH ", data.temperature / 100.0f,
			data.pressure / 100.0f, data.humidity / 1000.0f );
		/* Avoid using measurements from an unstable heating setup */
		if(data.status & BME680_GASM_VALID_MSK)
			printf(", G: %d ohms", data.gas_resistance);
		printf("\r\n");
	}
```

Note that the return value 2 stands for a warning, `BME680_W_NO_NEW_DATA` 

We will update the example in any upcoming update.

axelbusch, 2017-12-15:
Hi @BoschSensortec ,

thank you for your input!
No I get changing values for humidity, but not for temperature, pressure, and gas resistance:

> WR: 224, 1 0xb6
RD: 208, 1 0x61
RD: 137, 25 0x80 0x55 0x67 0x3 0x2f 0xfd 0x8f 0xd7 0xd6 0x58 0xff 0xc7 0x1f 0x90 0xff 0x3a 0x1e 0x0 0x0 0x8 0xee 0xfc 0xf8 0x1e 0x3
RD: 225, 16 0x40 0x19 0x2b 0x0 0x2d 0x14 0x78 0x9c 0x0 0x66 0x12 0xa7 0xea 0x12 0x71 0x0
RD: 2, 1 0x16
RD: 0, 1 0x32
RD: 4, 1 0xf3
WR: 90, 3 0x6c 0x64 0x65
RD: 116, 1 0x0
RD: 117, 1 0x7
RD: 116, 1 0x0
RD: 114, 1 0x2
RD: 113, 1 0x10
WR: 117, 7 0xb 0x74 0x8c 0x72 0x2 0x71 0x10
RD: 116, 1 0x0
WR: 116, 1 0x1
RD: 29, 15 0x80 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0x5f 0x80 0x0 0x0 0x0 0x20
T: 33.59 degC, P: 676.72 hPa, H 64.23 %rH , G: 12976914 ohms
WR: 90, 3 0x6c 0x64 0x65
RD: 116, 1 0x0
RD: 117, 1 0x7
RD: 116, 1 0x0
RD: 114, 1 0x2
RD: 113, 1 0x10
WR: 117, 7 0xb 0x74 0x8c 0x72 0x2 0x71 0x10
RD: 116, 1 0x0
WR: 116, 1 0x1
RD: 29, 15 0x80 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0x5c 0x80 0x0 0x0 0x0 0x20
T: 33.59 degC, P: 676.72 hPa, H 64.20 %rH , G: 12976914 ohms
WR: 90, 3 0x6c 0x64 0x65
RD: 116, 1 0x0
RD: 117, 1 0x7
RD: 116, 1 0x0
RD: 114, 1 0x2
RD: 113, 1 0x10
WR: 117, 7 0xb 0x74 0x8c 0x72 0x2 0x71 0x10
RD: 116, 1 0x0
WR: 116, 1 0x1
RD: 29, 15 0x80 0x0 0x80 0x0 0x0 0x80 0x0 0x0 0x58 0x5d 0x80 0x0 0x0 0x0 0x20
T: 33.59 degC, P: 676.72 hPa, H 64.21 %rH , G: 12976914 ohms
WR: 90, 3 0x6c 0x64 0x65
RD: 116, 1 0x0
RD: 117, 1 0x7
RD: 116, 1 0x0
RD: 114, 1 0x2
RD: 113, 1 0x10
WR: 117, 7 0xb 0x74 0x8c 0x72 0x2 0x71 0x10
RD: 116, 1 0x0
WR: 116, 1 0x1

BoschSensortec, 2017-12-15:
Hi @axelbusch,

There seems to be a problem with the write. Values are not being set into the registers. Can you check you verify if the write functions are working. This is evident in the log you just posted. Register 0x74  or 116d is unchanged and holding the default values. It should contain 0x8c, but during the next read, it contains 0x00.

WR: 117, 7 0xb **0x74 0x8c** 0x72 0x2 0x71 0x10
RD: 116, 1 **0x0**
WR: 116, 1 0x1

axelbusch, 2017-12-16:
Hi @BoschSensortec ,
finally I figured out why my read and write functions made some strange things (puh). Read after write did not result in the same values. Anyway, I managed to get them work. Thank you for your help so far.
Now I try to switch over to the precompiled lib to get air quality values. Unfortunately, I can't go any further with the init call. I got:

> Illegal instruction

I use the following code (main is from BSEC_1.4.5.1_Generic_Release_20171116/example/bsec_iot_example.c; I use the libalgobsec from BSEC_1.4.5.1_Generic_Release_20171116/algo/bin/RaspberryPI/PiZero_ArmV6-32bits/ since I try to run on raspi zero):
```
#include <stdio.h>
#include <stdlib.h>
#include <time.h>
#include <linux/i2c-dev.h>
#include <fcntl.h>
#include <string.h>
#include <sys/ioctl.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <unistd.h>
#include "bme680.h"
#include "bsec_integration.h"

int i2cDeviceHandle;

void i2cBegin(int addr)
{
	i2cDeviceHandle = open("/dev/i2c-1", O_RDWR);
	if (i2cDeviceHandle < 0) {
		perror("i2c Open failed");
		exit(1);
	}
	if (ioctl(i2cDeviceHandle, I2C_SLAVE, addr) < 0) {
		perror("i2c Set Address Failed");
		exit(1);
	}
}

void i2cClose()
{
	close(i2cDeviceHandle);
}

void user_delay_ms(uint32_t period)
{
    usleep(period*1000);
}

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

    uint8_t reg[1];
    reg[0]=reg_addr;

    if (write(i2cDeviceHandle, reg, 1) != 1) {
	rslt = -1;
    }
    if (read(i2cDeviceHandle, reg_data, len) < len) {
	rslt = -1;
    }

    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

    uint8_t reg[16];
    reg[0]=reg_addr;
	
    for (int idx = 1; idx < len + 1; idx++)
       reg[idx] = reg_data[idx - 1];

    if (write(i2cDeviceHandle, reg, len + 1) < len + 1) {
	rslt = -1;
        exit(1);
    }

    return rslt;
}

uint32_t state_load(uint8_t *state_buffer, uint32_t n_buffer)
{
	return 0;
}

void state_save(const uint8_t *state_buffer, uint32_t length)
{

}

uint32_t config_load(uint8_t *config_buffer, uint32_t n_buffer)
{
	return 0;
}

void output_ready(int64_t timestamp, float iaq, uint8_t iaq_accuracy, float temperature, float humidity,
		     float pressure, float raw_temperature, float raw_humidity, float gas, bsec_library_return_t bsec_status)
{
	printf("Output");
}

int64_t get_timestamp_us()
{
	    int64_t system_current_time = 0;
	    return system_current_time;
}
int main(int argc, char *argv[] )
{

	return_values_init ret;

	i2cBegin(BME680_I2C_ADDR_PRIMARY);

	ret = bsec_iot_init(BSEC_SAMPLE_RATE_LP, 0.0f, user_i2c_write, user_i2c_read, user_delay_ms, state_load, config_load);
	if (ret.bme680_status)
	{
	        /* Could not intialize BME680 */
	        return (int)ret.bme680_status;
	 }
	 else if (ret.bsec_status)
	 {
	     /* Could not intialize BSEC library */
	     return (int)ret.bsec_status;
	 }
	 bsec_iot_loop(user_delay_ms, get_timestamp_us, output_ready, state_save, 10000);

	i2cClose();
	
	return 0;
}
````
Do you have any idea? Thank you a lot!

BoschSensortec, 2017-12-18:
Hi @axelbusch,

It is not clear what throws the error from your comment. Is it the compiler? Or the init function itself? If it is the function, what is the error code it is throwing.

In addition, do you have your entire project hosted somewhere so that we can have a look at it, including makefiles and other dependencies?

alexh-name, 2017-12-27:
I wrote a working example for Raspberry Pi (Zero). Maybe it can help you: https://github.com/alexh-name/bsec_bme680_linux