BoschSensortec/BME680_driver #15: Save state/save config

damo1023, 2017-12-22:
Hello

I plan to use eeprom to save/load state and config and looking for size of those.

I assume config is 400 bytes and state appears to be 65 bytes long. Am I right or not? Does state change during longer periods of operation?

Thank you for reply.

BoschSensortec, 2018-01-10:
Hi @damo1023,

No the state does not change during operation but it might change from version to version of the BSEC library. We will have a separate constant for state size in the upcoming release.

damo1023, 2018-01-11:
Ok, thank you.

What do I do when I load state at startup? Is loading into struct enough or do I have to call some additional function? Shall I load state prior or after  calling function _bsec_iot_init_?

I also have a problem achieving IAQaccuracy>1. I put all variables to global and I am calling function strictly every 3000msec, but accuracy does not go more than 1. If I call your function _bsec_iot_loop_, accuracy goes higher. This problem gives me a headache for some time :|

Here is my function:

`void read_bme(sleep_fct sleep, get_timestamp_us_fct get_timestamp_us, output_ready_fct output_ready,
                    state_save_fct state_save, uint32_t save_intvl)
{
      
    time_stamp = get_timestamp_us() * 1000;                      // get the timestamp
    n_samples++;
     
    bsec_sensor_control(time_stamp, &sensor_settings);           // Retrieve sensor settings
    bme680_bsec_trigger_measurement(&sensor_settings, sleep);    // Trigger a measurement if necessary
        
    // Read data from last measurement
    num_bsec_inputs = 0;
    bme680_bsec_read_data(time_stamp, bsec_inputs, &num_bsec_inputs, sensor_settings.process_data);
        
    // perform the actual processing
    bme680_bsec_process_data(bsec_inputs, num_bsec_inputs, output_ready);

    if (n_samples == save_intvl) 
    {
      bsec_status = bsec_get_state(0, bsec_state, sizeof(bsec_state), work_buffer, sizeof(work_buffer), &bsec_state_len);
      if (bsec_status == BSEC_OK)   { state_save(bsec_state, bsec_state_len);
                                      n_samples=0;
                                    }
    }
} '

Thank you very much for your help!

Damjan

BoschSensortec, 2018-01-12:
Hi @damo1023,

In the upcoming release, we will add examples on showing how to use these features in a simpler manner.

damo1023, 2018-01-12:
Ok, thank you. Is new release comming soon?

BoschSensortec, 2018-01-12:
Hi @damo1023,

We cannot commit at this point of time with absolute certainty, but a little over a week.

closed 2018-03-28 by damo1023