BoschSensortec/BME680_driver #17: BSEC on Arduino Mega 2560

windkh, 2018-01-28:
I tried to get the latest BSEC library (BSEC_1.4.5.1_Generic_Release_20171214) running on an Arduino Mega 2560. Compiling, linking, wiring and communication between sensor and arduino via I2C already works. But BSEC returns an error code.  The serial output looks as follows
`
Opening port
Port open

bus_write dev=118 reg=224 len=1
bus_read  dev=118 reg=208 len=1
bus_read  dev=118 reg=137 len=25
bus_read  dev=118 reg=225 len=16
bus_read  dev=118 reg=2 len=1
bus_read  dev=118 reg=0 len=1
bus_read  dev=118 reg=4 len=1
Error while initializing BSEC library: 2560
`

The following code generates the error:
`
	return_values_init ret = bsec_iot_init(BSEC_SAMPLE_RATE_LP, 0.0f,
		bus_write,
		bus_read,
		sleep,
		loadState,
		loadConfig);
	if (ret.bme680_status)
	{
		/* Could not intialize BME680 */
		Serial.print("Error while initializing BME680: ");
		Serial.println(ret.bme680_status, DEC);
		return;
	}
	else if (ret.bsec_status)
	{
		/* Could not intialize BSEC library */
		Serial.print("Error while initializing BSEC library: ");
		Serial.println(ret.bsec_status, DEC);
		return;
	}
`

The bsec_status defined as enum, but I can not find the value 2560 in it:
`
typedef enum
{
    BSEC_OK = 0,                                    /*!< Function execution successful */
    BSEC_E_DOSTEPS_INVALIDINPUT = -1,               /*!< Input (physical) sensor id passed to bsec_do_steps() is not in the valid range or not valid for requested virtual sensor */
    BSEC_E_DOSTEPS_VALUELIMITS = -2,                /*!< Value of input (physical) sensor signal passed to bsec_do_steps() is not in the valid range */
    BSEC_E_DOSTEPS_DUPLICATEINPUT = -6,             /*!< Duplicate input (physical) sensor ids passed as input to bsec_do_steps() */
    BSEC_I_DOSTEPS_NOOUTPUTSRETURNABLE = 2,         /*!< No memory allocated to hold return values from bsec_do_steps(), i.e., n_outputs == 0 */
    BSEC_W_DOSTEPS_EXCESSOUTPUTS = 3,               /*!< Not enough memory allocated to hold return values from bsec_do_steps(), i.e., n_outputs < maximum number of requested output (virtual) sensors */
    BSEC_W_DOSTEPS_TSINTRADIFFOUTOFRANGE = 4,       /*!< Duplicate timestamps passed to bsec_do_steps() */
    BSEC_E_SU_WRONGDATARATE = -10,                  /*!< The sample_rate of the requested output (virtual) sensor passed to bsec_update_subscription() is zero */
    BSEC_E_SU_SAMPLERATELIMITS = -12,               /*!< The sample_rate of the requested output (virtual) sensor passed to bsec_update_subscription() does not match with the sampling rate allowed for that sensor */
    BSEC_E_SU_DUPLICATEGATE = -13,                  /*!< Duplicate output (virtual) sensor ids requested through bsec_update_subscription() */
    BSEC_E_SU_INVALIDSAMPLERATE = -14,              /*!< The sample_rate of the requested output (virtual) sensor passed to bsec_update_subscription() does not fall within the global minimum and maximum sampling rates  */
    BSEC_E_SU_GATECOUNTEXCEEDSARRAY = -15,          /*!< Not enough memory allocated to hold returned input (physical) sensor data from bsec_update_subscription(), i.e., n_required_sensor_settings < #BSEC_MAX_PHYSICAL_SENSOR */
    BSEC_E_SU_SAMPLINTVLINTEGERMULT = -16,          /*!< The sample_rate of the requested output (virtual) sensor passed to bsec_update_subscription() is not correct */
    BSEC_E_SU_MULTGASSAMPLINTVL = -17,              /*!< The sample_rate of the requested output (virtual), which requires the gas sensor, is not equal to the sample_rate that the gas sensor is being operated */
    BSEC_E_SU_HIGHHEATERONDURATION = -18,           /*!< The duration of one measurement is longer than the requested sampling interval */
    BSEC_W_SU_UNKNOWNOUTPUTGATE = 10,               /*!< Output (virtual) sensor id passed to bsec_update_subscription() is not in the valid range; e.g., n_requested_virtual_sensors > actual number of output (virtual) sensors requested */
    BSEC_I_SU_SUBSCRIBEDOUTPUTGATES = 12,           /*!< No output (virtual) sensor data were requested via bsec_update_subscription() */
    BSEC_E_PARSE_SECTIONEXCEEDSWORKBUFFER = -32,    /*!< n_work_buffer_size passed to bsec_set_[configuration/state]() not sufficient */
    BSEC_E_CONFIG_FAIL = -33,                       /*!< Configuration failed */
    BSEC_E_CONFIG_VERSIONMISMATCH = -34,            /*!< Version encoded in serialized_[settings/state] passed to bsec_set_[configuration/state]() does not match with current version */
    BSEC_E_CONFIG_FEATUREMISMATCH = -35,            /*!< Enabled features encoded in serialized_[settings/state] passed to bsec_set_[configuration/state]() does not match with current library implementation */
    BSEC_E_CONFIG_CRCMISMATCH = -36,                /*!< serialized_[settings/state] passed to bsec_set_[configuration/state]() is corrupted */
    BSEC_E_CONFIG_EMPTY = -37,                      /*!< n_serialized_[settings/state] passed to bsec_set_[configuration/state]() is to short to be valid */
    BSEC_E_CONFIG_INSUFFICIENTWORKBUFFER = -38,     /*!< Provided work_buffer is not large enough to hold the desired string */
    BSEC_E_CONFIG_INVALIDSTRINGSIZE = -40,          /*!< String size encoded in configuration/state strings passed to bsec_set_[configuration/state]() does not match with the actual string size n_serialized_[settings/state] passed to these functions */
    BSEC_E_CONFIG_INSUFFICIENTBUFFER = -41,         /*!< String buffer nsufficient to hold serialized data from BSEC library */
    BSEC_E_SET_INVALIDCHANNELIDENTIFIER = -100,     /*!< Internal error code */
    BSEC_E_SET_INVALIDLENGTH = -104,                /*!< Internal error code */
    BSEC_W_SC_CALL_TIMING_VIOLATION = 100,          /*!< Difference between actual and defined sampling intervals of bsec_sensor_control() greater than allowed */
} bsec_library_return_t;
`

Any ideas what this error means?
Thanks in advance.

BoschSensortec, 2018-01-29:
Hi @windkh,

It might be a ```BSEC_W_SC_CALL_TIMING_VIOLATION``` with error code ```100```, and some type casting gone wrong which would explain the 100 turning into 2560. But it is still not definitive how that would be possible. Let us try a similar setup and get back to you.

windkh, 2018-01-29:
I can send you my sketch project as zip if it would help

BoschSensortec, 2018-01-29:
Hi @windkh,

Would help if you have changed some code snippets from the original.

windkh, 2018-01-29:
I sent two zip files. You can open the solution with visual studio 2015 (with visual micro) with Arduino 1.8 environment.

rpaa, 2018-02-09:
Having the exact same problem with Arduiono bsec example, using Arduino Mega 2560.
I can use bme680 with my own test Arduino project (based on Adafruit example) but bsec_iot_example fails in bsec_iot_init() at bsec_init() call with error code 2560.

My best quess is that AVR8_megaAVR version of bsec library checks the controller type and fails with error 2560 on Atmega 2560.  Is it so?

DanieleSgroi, 2018-03-06:
Hallo.
Same problem here: bsec_iot_init() returns error code 2560.

Is there any news on this issue?

thanks

DanieleSgroi, 2018-03-26:
Just tried with BSEC 1.4.6.0, different error code but same behavior for both standard and lite version of the new library.

Furthermore, I've verified that the error is raised when calling bsec_init() from inside bsec_iot_init() function.

Surprisingly enough, the bsec_get_version() function returns corerrect values, so some code should work inside the binary...

Is there somoene from Bosch that can explain and help?

Thanks...

BoschSensortec, 2018-03-26:
Hi @DanieleSgroi,

We have observed the same and are currently investigating it. It seems to happen only in the Arduino IDE. The return of the `bsec_init` is hardcoded to `BSEC_OK`. It seems to be some issue with the linking with the Arduino IDE's settings.

ashishshukla78, 2018-03-26:
I am facing similar problem where bsec_init returns 2304 on Arduino Mega. 
Thanks

BoschSensortec, 2018-03-28:
Hi @ashishshukla78 , @DanieleSgroi , @windkh, @rpaa,

It is a case of type-misalignment. The BSEC library was compiled with the option `-fshort_enums`  which is not the case with the code compiled in the Arduino IDE. We will update the release without this build option.

ashishshukla78, 2018-04-09:
Is there a plan to fix this?

DanieleSgroi, 2018-04-10:
@BoschSensortec 
any estimates for the new binary?

ashishshukla78, 2018-04-11:
Thanks DanieleSgroi! It is working for me as well.