BoschSensortec/BME680_driver #21: Ambient Temperature

agamagtdM10102148, 2018-02-14:
@BoschSensortec 
"**amb_temp**"  of  "**struct bme680_dev**" in **bme680_def.h** is used in **calc_heater_res()**.
amb_temp is provided in **bme680_self_test()** of bme680_selftest.c while it is not used in **README.md**.

Does that matter for _temperature accuracy_ used in **_BSEC_** ?

Thank you so much.

BoschSensortec, 2018-02-14:
@agamagtdM10102148,

It does affect the set heater temperature slightly. Not the temperature sensor's accuracy. We will highlight this in the README.

mickpf, 2018-02-18:
Dear Sirs,
As I have stated, "amb_temp" is used in "calc_heater_res" without any value assigned to this variable!
Would it not be correct to perform a value assignment in the "read_field_data" function after "data-> temperature" was calculated?

Kind Regards,
Michael Wachs

BTW: The datasheet of BME680 lacks any mention / description of the parameter registers t1-t3, p1-p10, h1-h7 and gh1-gh3. Have they been declared as a secret? Are there any further register descriptions missing?

BoschSensortec, 2018-02-19:
Hi @mickpf,

Typically this is set by performing a few temperature readings without operating the gas sensor, or simply set to 25 prior to configuring the sensor. This will be updated in the README in an upcoming release.

Updates to the datasheet are underway.