BoschSensortec/BME680_driver #22: BSEC lib

Robertor111, 2018-02-15:
Hello,

I'm trying to implement the bsec to my project. I'm using the the pre-compiled libalgobsec.lib file. The version read out for bsec works fine 1.4.5.1.  But when colling function bsec_init(), the system crashes, due to hardfault (INVALIDSTATE), any ideas what's could be the reason?

Best regards

BoschSensortec, 2018-02-15:
Hi @Robertor111,

Can you describe a little bit about your setup, hardware, compiler and connections? Also a build log would be helpful.

BoschSensortec, 2018-02-15:
Hi @Robertor111,

Do you have a build log?

Robertor111, 2018-02-15:
*** Using Compiler 'V5.06 update 4 (build 422)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Rebuild target 'BME680_USB2'
assembling startup_stm32l476xx.s...
compiling usbd_desc.c...
compiling system_stm32l4xx.c...
compiling usbd_cdc_if.c...
compiling stm32l4xx_hal_msp.c...
compiling usbd_conf.c...
compiling stm32l4xx_it.c...
compiling usb_device.c...
compiling main.c...
compiling bme680.c...
..\Src\bme680.c(852): warning:  #61-D: integer operation result is out of range
  	var4 = (1 << 31);
..\Src\bme680.c: 1 warning, 0 errors
compiling bsec_integration.c...
..\Src\bsec_integration.c(161): warning:  #177-D: variable "bsec_status"  was declared but never referenced
      bsec_library_return_t bsec_status = BSEC_OK;
..\Src\bsec_integration.c(238): warning:  #550-D: variable "bme680_status"  was set but never used
    int8_t bme680_status = BME680_OK;
..\Src\bsec_integration.c(298): warning:  #550-D: variable "bme680_status"  was set but never used
      int8_t bme680_status = BME680_OK;
..\Src\bsec_integration.c: 3 warnings, 0 errors
compiling bsec_serialized_configurations_iaq.c...
compiling usbd_ioreq.c...
compiling bsec_iot_example.c...
compiling usbd_cdc.c...
compiling usbd_ctlreq.c...
compiling usbd_core.c...
compiling stm32l4xx_ll_usb.c...
compiling stm32l4xx_hal_flash_ex.c...
compiling stm32l4xx_hal_i2c.c...
compiling stm32l4xx_hal_cortex.c...
compiling stm32l4xx_hal_rtc.c...
compiling stm32l4xx_hal_tim.c...
compiling stm32l4xx_hal_dma.c...
compiling stm32l4xx_hal_pwr_ex.c...
compiling stm32l4xx_hal_pcd.c...
compiling stm32l4xx_hal_dma_ex.c...
compiling stm32l4xx_hal_rcc.c...
compiling stm32l4xx_hal_rtc_ex.c...
compiling stm32l4xx_hal_gpio.c...
compiling stm32l4xx_hal.c...
compiling stm32l4xx_hal_rcc_ex.c...
compiling stm32l4xx_hal_flash_ramfunc.c...
compiling stm32l4xx_hal_i2c_ex.c...
compiling stm32l4xx_hal_tim_ex.c...
compiling stm32l4xx_hal_pwr.c...
compiling stm32l4xx_hal_pcd_ex.c...
compiling stm32l4xx_hal_flash.c...
linking...
Program Size: Code=34320 RO-data=2032 RW-data=376 ZI-data=9360  
"BME680_USB2\BME680_USB2.axf" - 0 Error(s), 4 Warning(s).

BoschSensortec, 2018-02-15:
Hi @Robertor111,

We are looking specifically for the build options. C flags, and other ARMCC options that are enabled. Both for compiling and linking.

Robertor111, 2018-02-15:
[image]

ESP32DE, 2018-02-18:
@Robertor111 

Overflow!
> ..\Src\bme680.c(852): warning: #61-D: integer operation result is out of range

see 
https://github.com/BoschSensortec/BME680_driver/pull/16

change the code line

`var4 = (int32_t) (1U << 31);`

the other 3 warnings in your compile is simple unused vars.
hope this helps

@BoschSensortec 
when you use my PR ?
should i close it `?

für mich sehr unverständlich meinen code vorschlag nicht nutzen zu wollen!

ESP32DE, 2018-02-18:
so closed my PR now
@BoschSensortec  why you use Github if you not use PR from outside`?
....

Da hat die Verwendung des BME680 für mich gleich wieder einen Beigeschack.

ESP32DE, 2018-02-18:
@BoschSensortec 
cause BSEC_1.4.5.1_Generic_Release_20171214 LIB+API also affected
i [reopen the PR](https://github.com/BoschSensortec/BME680_driver/pull/16)

Robertor111, 2018-02-19:
I have also corrected the (#61-D integer operation result is out of range), but it still crashes. 

The only thing that i changed to the code, is in file bsec_iot_example.c:

 #define BME680_WRITE_ADD 0xEE
 #define BME680_READ_ADD 0xEF
 
 
int8_t bus_write(uint8_t dev_addr, uint8_t reg_addr, uint8_t *reg_data_ptr, uint16_t data_len)
{
  HAL_I2C_Mem_Write(&hi2c1, BME680_WRITE_ADD, reg_addr,I2C_MEMADD_SIZE_8BIT, reg_data_ptr, data_len, 0xFFFF);
	  
    return 0;
}

int8_t bus_read(uint8_t dev_addr, uint8_t reg_addr, uint8_t *reg_data_ptr, uint16_t data_len)
{

	  HAL_I2C_Mem_Read(&hi2c1, BME680_READ_ADD, reg_addr, I2C_MEMADD_SIZE_8BIT, reg_data_ptr, data_len, 0xFFFF);
	
    return 0;
}

void sleep(uint32_t t_ms)
{
	HAL_Delay(t_ms);
}

BoschSensortec, 2018-02-19:
@Robertor111,

The device address should be ``` BME680_I2C_ADDR_PRIMARY ``` or ``` BME680_I2C_ADDR_SECONDARY ``` for both read and write. Is there any reason you are using it differently? In certain HALs, due to the implementation being in 8bit I2C addressing, you will need to shift the address to the left by 1.

Robertor111, 2018-02-19:
First i implemented basic comunication to the Bme680, to test the proper working of the sensor. After that i tried to implement the bsec to my project, to calculate IAQ, where the problem started. The init and communication to the sensor goes well, and not using BME680_I2C_ADDR_PRIMARY or BME680_I2C_ADDR_SECONDARY should not efect the program work flow, since i have entered the i2c address directly and it works. 

This is the part where it stops working:

[image]

Robertor111, 2018-02-19:
[image]

BoschSensortec, 2018-02-19:
Hi @Robertor111,

It seems like an issue with the linking the compiled library.

You will have to fill in the [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form) and one of our FAEs will get in touch with you.

ESP32DE, 2018-02-19:
@Robertor111 
which IDE ,  Debug Tools, Toolset you use, so i can start here with the same and try to help you over this Public Github. i think, it would be very good, if you keep us here updated from your founds, cause there are more people with the same problematic.

@BoschSensortec 
...

Robertor111, 2018-02-19:
Hi,
I have already fulfill the contact form, before posting to a forum. But I still did not get the answer.

Robertor111, 2018-02-21:
Hi,
the files provided with bsec include also a config file for different power modes, but i can't see any linkage to them. could dis couse the mentioned problem?

Robertor111, 2018-03-30:
Hi,

the BSEC initialisation for cortex M4 should be fiksed in realise: 1.4.6.0, but still geting the same result. 
The version of  read out give me 1.3.4.1.????! Could some one conferme that the labrary for cortex M4 works, before spending more time on it.

Best Regards

BoschSensortec, 2018-04-03:
Hi @Robertor111,

It seems like you are linking the wrong binary. Can you check your build logs to see where the `libalgobsec.a` is used from and whether this is from the latest release? Also, replace the headers, `bsec_datatypes.h` and `bsec_interface.h`.

phanirajkiran, 2018-05-28:
Hi Robertor
Can you please share your source(zip) to analyze the root cause. 
Thanks
Phani

phanirajkiran, 2018-06-01:
Hi Bosch
With the latest Source which got posted on Feb14 with  commit e6b9bba. Leads to Lib linking issue. 
Suppose i made VFP register linking using  Floating Point ABI = Soft, and proceed further. When 
bsec_init() when called its leading to the STM32L4R5ZI which is of M4 to a MCU RESET. 

Please release a Binary which can be linked using **hard**
Please find the thread for STM32R5ZI and BME680 Lib MCU reset.
**https://community.st.com/thread/50397-stm32l4r5zi-reset-issue**

Thanks
Phani

aduecho, 2018-09-26:
@Robertor111 Hi, I get the same problem recently, finally I found it's caused by heap overflow. now mine works fine. hopefully this would help you a little bit