BoschSensortec/BME680_driver #25: Sometimes BME680 stays in "BME680_FORCED_MODE" indefinitely

easySolar, 2018-02-18:
Hi,

We are developing an application using the BSEC library (BSEC_1.4.5.1_Generic_Release_20171214) running on an ESP32 with Arduino IDE.
Everything works well (compiling, linking, wiring, communication between BME680 and arduino via I2C) but BSEC sometimes enters an endless loop in file "bsec_integration.c" / function "bme680_bsec_trigger_measurement(bsec_bme_settings_t *sensor_settings, sleep_fct sleep)":

    /* Call the API to get current operation mode of the sensor */
    bme680_status = bme680_get_sensor_mode(&bme680_g);  
    /* When the measurement is completed and data is ready for reading, the sensor must be in BME680_SLEEP_MODE.
     * Read operation mode to check whether measurement is completely done and wait until the sensor is no more
     * in BME680_FORCED_MODE. */
    while (bme680_g.power_mode == BME680_FORCED_MODE)
    {
        /* sleep for 5 ms */
        sleep(5);
        bme680_status = bme680_get_sensor_mode(&bme680_g);
    }
	
Sometimes "bme680_g.power_mode" stay in "BME680_FORCED_MODE" indefinitely.

It can happen after a few minutes or after several hours or days (We have run the test for more than two days without problem). We tested two different versions of ESP2 cards with two BME680 Adafruit cards and two BOSCH SHUTTLE BOARD DEV KIT BME680.
The problem occurs randomly and when the application is restarted by a hardware reset - it is not necessary to make a power off - it restarts for several hours before hanging again.

Do you have an idea?
Thank you if you can help!

BoschSensortec, 2018-02-19:
@easySolar,

Have you verified with a logic analyzer that the I2C/SPI driver is actually reading data from the sensor and not providing the last read buffer? Can you also monitor the values on the bus?

easySolar, 2018-02-19:
Hello,

Yes we are going to do that. I comme back to you asap, it could take from a few hours to a few day before the problem arise.

Thank you

agamagtdM10102148, 2018-03-02:
@BoschSensortec  @easySolar 
bme680_status = bme680_get_sensor_mode(&bme680_g);
The above operation is done independent of  if (sensor_settings->trigger_measurement).
Does that matter ? (Although my program ran correctly.)

agamagtdM10102148, 2018-03-02:
@easySolar 
        bme680_status = bme680_set_sensor_settings(set_required_settings, &bme680_g);
        bme680_status = bme680_set_sensor_mode(&bme680_g);
After the above two operations in bme680_bsec_trigger_measurement() are executed, BME680 must be in sleeping mode.
Due to  this line "dev->power_mode = BME680_SLEEP_MODE;" in bme680_set_sensor_settings();

BoschSensortec, 2018-03-02:
@agamagtdM10102148 & @easySolar,

The sensor goes back to sleep mode after the measurement is complete. Hence reading the power mode back will usually result in BME680_SLEEP_MODE, especially if the measurement duration is complete.

Before triggering the forced mode again, the driver has to make sure that the sensor is in sleep mode or else the command will be ignored.

kylix34, 2018-04-23:
Hi! Is it possible to download BSEC_1.4.5.1_Generic_Release_20171214 from somewhere?

BoschSensortec, 2018-04-24:
Hi @kylix34,

The older revisions are currently unavailable via our [website](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form). You can use the contact form to request for the same.