BoschSensortec/BME680_driver #27: Error while initializing BME680

sisamiwe, 2018-03-01:
Hi,
i'm trying to use an BME680 from watterott on an ESP8266 (Wemos D1 mini). 
First I want to use your example "bsec_iot_example" as is. All the needed files are copied.
Compiling and flashing works, but on serial output  I always get "Error while initializing BME680".
The return of the function "ret.bme680_status" is "-2". What does it mean? How can I get that working?
Thank you!

wklenk, 2018-03-01:
Probably you made a mistake with the wiring. It can be hard to get it right, I also spent much time to find the reasons for problems like this. My recommendation: Use I2C to talk to the BME680 board, and start with the I2C Scanner sketch described here: https://playground.arduino.cc/Main/I2cScanner
It tries out all I2C adresses and should find the BME680. If it doesn't, something is wrong with your wiring. This I2C Scanner will also tell you to which I2C address the BME680 is listening to: Either to 0x76 (constant: BME680_I2C_ADDR_PRIMARY) or 0x77 (constant:  BME680_I2C_ADDR_SECONDARY). This is also a source of error if you don't exactly know which address is configured to use.

sisamiwe, 2018-03-01:
The sensor is connected to 0x77. It has been found be the I2C Scanner.
But the "bsec_iot_example" does have the same I2C scanner logic, or?