BoschSensortec/BME680_driver #28: BSEC Outputs: (un)scaled IAQ, tVOC, and eCO2

agamagtdM10102148, 2018-03-02:
@BoschSensortec 
We have tested BSEC with additional hidden properties, such as unscaled IAQ, tVOC, as well as eCO2 and found that
1. unscaled IAQ is not affected by its ambient environment so seriously as scaled IAQ is.
2. As BSEC is in not in run_in state (run_in == 0), tVOC and CO2 reads 500 ppb and 221 ppm, respectively. As BSEC is in run_in state (run_in == 1), tVOC goes around and up to 0.6 ppm (600 ppb)  and eCO2 goes around 220 ppm in my office. And IAQ is below 49 (GOOD air quality).
3. When compared with Sensirion SGP30, SGP30's  (tVOC , eCO2) is (<140, 400~1000) in my office (GOOD air quality).
Although air quality classifications for both SGP30 and BME680 are the same, BME680's tVOC is much higher than SGP30's tVOC.  May BoschSensortec help us to correct BME680's tVOC and eCO2 readings ?

BoschSensortec, 2018-03-02:
@agamagtdM10102148,

Kindly address your questions to your local FAE. If one hasn't already been assigned, then kindly put in a request via our contact form. Deleting your previous comment due to violations of the NDA.

agamagtdM10102148, 2018-03-02:
@BoschSensortec 
Thanks