BoschSensortec/BME680_driver #29: Filter settings for temperature and pressure

agamagtdM10102148, 2018-03-02:
@BoschSensortec 
In selftest.c, no filter settings are used. While in README.md,  a filter setting are used.
My question is:
1. the purpose of filter
2. how to choose suitable filter size

Thanks a lot.

BoschSensortec, 2018-03-02:
Hi @agamagtdM10102148,

Kindly refer chapter **3.3.4 IIR Filter** in the [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME680-DS001-00.pdf).

agamagtdM10102148, 2018-03-08:
@BoschSensortec
Thank you.
The purpose can be understand by referring to datasheet while there is no idea to choose the proper filter size for certain application or environment.

BoschSensortec, 2018-03-08:
Hi @agamagtdM10102148,

We do not have application notes for the BME680 regarding filter and oversampling settings. We will get back to you when one is available.

Thank you for your feedback.