BoschSensortec/BME680_driver #30: Support for multiple physical devices and thread safety

fivdi, 2018-03-07:
I'd like to implement a Node.js module for the BME680 and have a few questions.

Is it correct that both the BME680_driver and the BSEC fusion library can support multiple physical devices at the same time in the same application?

For example, if I2C is being used, there could be one device at address 0x76 and another at 0x77 and everything should function correctly?

The sample code in `bsec_integration.c` that's provided as part of the BSEC fusion library doesn't appear to have been implemented to handle multiple physical devices at the same time, so if multiple physical devices need to be handled this code would need to be refactored, correct?

Is the code in BME680_driver and the BSEC fusion library thread safe?

BoschSensortec, 2018-03-07:
Hi @fivdi,

The BME680 driver can support multiple sensors, but the BSEC fusion library currently available only supports 1 instance. There is a multi-instance version available upon request via our contact form. In the future we plan to offer the multi-instance version as part of the standard release too.

Both the driver and the library are thread safe.

fivdi, 2018-03-07:
@BoschSensortec Thanks for the information.

> In the future we plan to offer the multi-instance version as part of the standard release too.

This would be good.

closed 2018-10-20 by fivdi