BoschSensortec/BME680_driver #31: Using the BSEC fusion library in a shared library

fivdi, 2018-03-08:
I'm currently attempting to implement a Node.js addon for the BME680 that can be used on Linux boards like the Raspberry Pi, BeagleBone Black or C.H.I.P. The goal is for the addon to be able to use code from the BME680 driver and the BSEC fusion library.

A Node.js addon is a shared library not an executable program. The executable program is `node` itself and it dynamically loads addons upon request at runtime.

I intended adding the content of the BSEC fusion library to the Node.js addon at install/compile time with compiler options `-Wl,-whole-archive /usr/lib/libalgobsec.a -Wl,-no-whole-archive`. However, it looks like this isn't going to work as the libalgobsec.a static library doesn't appear to have been compiled with the `-fPIC` option. The error message mentioning `-fPIC` can be seen below if you scroll all the way to the right.

Is it possible that the BSEC fusion library which is a static library is not suitable for usage in a shared library like a Node.js addon?

```
> bme680-sensor@0.0.0 install /home/pi/bme680-sensor
> node-gyp rebuild

make: Entering directory '/home/pi/bme680-sensor/build'
  SOLINK_MODULE(target) Release/obj.target/node_modules/node-addon-api/src/nothing.node
  COPY Release/nothing.node
  CC(target) Release/obj.target/addon/src/bme680.o
  CC(target) Release/obj.target/addon/src/bsec_integration.o
../src/bsec_integration.c: In function ‘bsec_iot_init’:
../src/bsec_integration.c:161:27: warning: unused variable ‘bsec_status’ [-Wunused-variable]
     bsec_library_return_t bsec_status = BSEC_OK;
                           ^~~~~~~~~~~
../src/bsec_integration.c: In function ‘bme680_bsec_trigger_measurement’:
../src/bsec_integration.c:237:12: warning: variable ‘bme680_status’ set but not used [-Wunused-but-set-variable]
     int8_t bme680_status = BME680_OK;
            ^~~~~~~~~~~~~
../src/bsec_integration.c: In function ‘bme680_bsec_read_data’:
../src/bsec_integration.c:297:12: warning: variable ‘bme680_status’ set but not used [-Wunused-but-set-variable]
     int8_t bme680_status = BME680_OK;
            ^~~~~~~~~~~~~
  CC(target) Release/obj.target/addon/src/bsec_iot_example.o
  CXX(target) Release/obj.target/addon/src/addon.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
/usr/bin/ld: /usr/lib/libalgobsec.a(Bsec.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/libalgobsec.a(Bsec.o): error adding symbols: Bad value
collect2: error: ld returned 1 exit status
addon.target.mk:137: recipe for target 'Release/obj.target/addon.node' failed
make: *** [Release/obj.target/addon.node] Error 1
make: Leaving directory '/home/pi/bme680-sensor/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/opt/node-v8.9.4-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.9.59-v7+
gyp ERR! command "/home/pi/opt/node-v8.9.4-linux-armv7l/bin/node" "/home/pi/opt/node-v8.9.4-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/bme680-sensor
gyp ERR! node -v v8.9.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bme680-sensor@0.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bme680-sensor@0.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-03-08T19_44_58_219Z-debug.log
```

BoschSensortec, 2018-03-09:
@fivdi,

We will get back to your shortly with a -fPIC version.

fivdi, 2018-03-09:
Excellent.

Have you considered providing libalgobsec as a shared library?

BoschSensortec, 2018-03-09:
Hi @fivdi,

We didn't anticipate the use of BSEC as a shared library :) Nevertheless this is valuable input and we are working on it.

TuxTroll, 2018-05-15:
It would be nice, to get the library for Odroids :) (XU4)..

fivdi, 2018-10-20:
To workaround this issue a C program was implemented to communicate with the BME680. The Node.js program then communicated with the C program to access information from the BME680.

closed 2018-10-20 by fivdi