BoschSensortec/BME680_driver #33: BSEC "bsec_iot_once" instead of "bsec_iot_loop" for arduino

janzieg, 2018-03-24:
Hi there,

the interface of bsec integration is quite limited. There is an init- and loop function and when in use, the arduino loop function becomes useless as the bsec while-loop blocks processing.

It would be nice to also have something like "bsec_iot_once" for single time measurements that could be used with the arduino main loop function (and so with other operations).

(i added my own function to bsec_integration right now for this as a copy from bsec_iot_loop but without while-loop and changed state saving behavior).

What do you think about it ?

BoschSensortec, 2018-03-26:
Hi @janzieg,

We will be releasing a Arduino friendly library based on the existing BSEC library shortly.

janzieg, 2018-03-26:
@BoschSensortec Thx, looking forward to it...

jttaylor86, 2018-04-04:
Hello, when is the Arduino library coming?

r0ehre, 2018-04-08:
Really looking forward to the release of the BSEC library for Arduino!

janzieg, 2018-04-13:
@BoschSensortec Thx, this works for me - I can now use it properly within the arduino loop function.

New library, new question, https://github.com/BoschSensortec/BSEC-Arduino-library/issues/5 :)

closed 2018-04-13 by janzieg