BoschSensortec/BME680_driver #37: [SAMB11 | Cortex-M0] BSEC integration problem

Bossmeister, 2018-04-06:
Hello,

I'm working on building an air quality system, for which I use an SAMB11 Xplained Pro board and the BME680. However, before using the SAMB11 I used a BBC micro:bit. Since I had ram limitations, I couldn't use Bluetooth at the same time. Thinking I got it to work on my BBC micro:bit with not more than 16kB of ram, I am now suprised I get an ram overflow with my SAMB11. When I compile in Atmel Studio 7 with all drivers included (I2C, Timers, Bluetooth), my memory data usage is around 9%. However, as soon as I include the BSEC lite version under "Librarys" and adding the BME680 API, my ram overflows. I have around 96kB of ram to use for application code. On your site it is stated that around 2kB of RAM is used for the libalgo, so I'm quite confused what is going wrong.

Maybe someone can point me in the right direction!

Kind regards

BoschSensortec, 2018-04-06:
@Bossmeister, 

We will be releasing a fix for the bsec_integration.c error reported for BSEC Lite.

Regarding the issue with the RAM overflow, it is hard to understand what the problem might be. Can you confirm that the ROM section of the algorithm is going into the flash?

cwq9999, 2018-04-06:
What is the expected timeline for the release of the fix of bsec_integration.c error for BSEC Lite?