BoschSensortec/BME680_driver #38: Parallel and sequential mode  support?

ourairquality, 2018-04-07:
There is discussion around about a 'parallel' and 'sequential' mode, but these do not appear to be documented in the data sheet. Is there any documentation for these?

Here is all I have seen in various drivers:

*  It seems parallel mode is mode 2, and sequential mode is mode 3. In sequential mode the heater runs only before a gas measurement, and in parallel mode the heater might run all the time??

* In sequential mode the sensor enters a standby mode for the period defined in register 0x6e bits 0 to 6, with a similar scale factor to the gas_wait times.

Would like to know if a sequence of gas heating profiles can be run in these modes. Can it cycle through different heating resistance and time set points?

Or are these considered redundant. When gas sampling is it just far more flexible to implement all the logic in a mcu and trigger each measurement as needed? Perhaps this would be so for the sequential mode, but does the parallel mode open new opportunities in gas measurement?

ourairquality, 2018-05-14:
Ping? Could really use some driver support for these modes that do not appear in the data sheet. I have started exploring temperature controlled modes of operation, but by using the 'forced' mode of operation. Would really like to be able to have the bme680 time these. It seems very helpful to measure the rate of change of the resistance in a non-equilibrium state - it might make the measurements much less sensitive to drift and more sensitive and perhaps even more selective. For example as describe in https://www.researchgate.net/publication/324253822_Highly_sensitive_benzene_detection_with_metal_oxide_semiconductor_gas_sensors_-_An_inter-laboratory_comparison

My own quick attempt does seem very promising. The rate of change seems much less sensitive to drift. However if there is a timing difference then that obviously causes an error in the rate. Perhaps that could be compensated for by timing the measurement times, but if the bme680 can time a schedule of temperatures and gas measurements then that might be better.

There is some code comments here https://os.mbed.com/users/UHSLMarcus/code/BME680/ that appears to give much more information than in the data sheet I have. Could this support be added to the official driver too?

Btw The discontinuities in the ADC output cause steps in the differentials, not nice, but that looks like a problem that can be solved.

I am trying to make this bme680 fly, and could use a little help?

BoschSensortec, 2018-05-14:
@ourairquality,

Can you share your contact email address. We will get in touch with you directly.

ourairquality, 2018-05-17:
@BoschSensortec Thank you, email is info @ ourairquality.org

Fwiw I appear to have the sequential mode working now too. To obtain some data, the gas schedule is programmed with 10 settings. Could not get the last to be used, so only 9. Set the first to 350 deg C for 200ms, then sample at 100 deg at increasing times, and for the last the heater register is set to zero and with a delay of a few seconds to cool between sets. The data appears on first inspection to be far less noisy than using the forced mode, see the attached plots. The papers talk of the derivative of the log of the conductance in the relaxation after the high to low temperature transition being roughly proportional to the concentration. The first plot is the log of the conductances for the each of the samples in the sequence, the highest is for the high temperature gas measurement and the lowest is the first low temperature sample, and the rest are the relaxation from that. The second plot is the initial derivative of that relaxation. The time unit is usec. The derivative does appear to respond in a similar manner to the variation in the gas samples. The discontinuities in the ADC have not been addressed here, can be seen, but please ignore that noise. I have other data that looked promising in terms of being less sensitive to drift, at least start up drift, but this short impulse does not show that. So I appear to have been able to gather some more data and cleaner data for now to explore this further.

[image]
[image]

It appeared to be necessary to define one more element in the sequence than is actually need, and not possible to use all 10 of the elements at once, so perhaps I am missing something there?

I have set one of the heater settings to zero, to attempt to have a defined cooling period between sets, and is that a good approach?

The derivative of the first sample appeared to give the cleanest signal. A final sequence might not measure the relaxation in such detail. Some papers describe a cascade of non-equilibrium samples, e.g. for  tea odour characterisation, and that might be possible too.

Each sample is a THPG measurement. This would appear to introduce a fall to ambient during the THP measurements when using the sequential mode, a somewhat different profile to that described in papers. If there are some more settings to play with then that might be useful to know?

I tried the 'parallel' mode too, got it generating some data, but it appears that the gas measurements were infrequent and I could not overcome that, there were a lot of THP measurements between the gas measurements whereas what appears to be need for exploring non-equilibrium states for VOC measurement is fast gas measurements. Perhaps there is more to understand there?

Please keep in mind that I work in the public and any measurement methods need to be open to scrutiny and reproducible.

ourairquality, 2018-05-17:
Seems to require an NDA.

closed 2018-05-17 by ourairquality