BoschSensortec/BME680_driver #42: Your examples are missing one important fact.

akirsch76, 2018-04-09:
Hello,

I tried to use your sensor with a raspberry p zero. So I compiled your examples. After that I was wondering why I always got the same sensor readings. After a few hours and a look in the sensor datasheet I found out that the sensor goes into sleep mode after measurement. But your example "Example for reading all sensor data" does not set the sensor in force mode in the while loop. So you get always the same sensor data. It would be nice if you could fix your sample! Thank you!

Kind regards

Alex

akirsch76, 2018-04-09:
Thank you very much! My little Programm runs now. But I didn't used the bsec library yet that does the mode switch invocation in the bsec code. I will switch my program to use bsec too.

iiv3, 2018-04-15:
It seems that b619094dfe507034e1034884f4638a78dae7646c has fixed the example.