BoschSensortec/BME680_driver #44: Integer pressure calculation lacks precision

ourairquality, 2018-04-15:
The 32 bit integer function `calc_pressure()` is rather inaccurate and an error of 5 Pa is not uncommon. Surely an integer version could do better, even if it had to use 64 bit integer operations in some places.

The most significant problem seems to be the limited precision of `var1` in the division at:
```pressure_comp = ((pressure_comp << 1) / (uint32_t)var1);```

Would a patch be accepted that used 64 bit integer operations where necessary to reduce the error?

BoschSensortec, 2018-04-16:
Hi @ourairquality,

Are you referring to this [line of code](https://github.com/BoschSensortec/BME680_driver/blob/b619094dfe507034e1034884f4638a78dae7646c/bme680.c#L918) specifically? Or are you referring to the whole API of `calc_pressure`.

We currently do not accept patches, but we do solve issues if reported and release it on Github.

BoschSensortec, 2018-04-17:
Hi @ourairquality,

Thank you for your feedback. We will look into this and update it as per your suggestions.

closed 2018-05-17 by ourairquality