BoschSensortec/BME680_driver #49: BSEC Air Quality Accuracy always 1

ukicar, 2018-04-23:
Hello,

How long do I have to wait to improve IAQ accuracy? After few minutes accuracy passes from 0 to 1 but it never goes further.
I'm using generic configuration profile for 3.3V, 3s and 4days. Does this mean that after 4 days accuracy will improve?
I'm also using the same code that was given in BSEC example bsec_ios_example.c
My bsec init is 
reta = bsec_iot_init(BSEC_SAMPLE_RATE_LP, 0.0f, bus_write, bus_read, sleep,
          state_load, config_load);

My platform is based on nrf51822 and using i2c bme680

BoschSensortec, 2018-04-24:
Hi @ukicar,

You need to expose the sensor to different qualities of air, it will then learn based on that input and then increment the accuracy level. 

One quick hack is to introduce the sensor to a strong source of VOCs like a white board marker for a few seconds until the sensor detects it and the IAQ readings saturate. You could additionally breathe on the sensor after a few minutes.

ukicar, 2018-04-24:
Yes exactly, it worked!!! Thanks for advice

closed 2018-04-24 by ukicar

BoschSensortec, 2018-04-24:
@ukicar,

We have a new repository for the [Arduino library](https://github.com/BoschSensortec/BSEC-Arduino-library).