BoschSensortec/BME680_driver #50: BSEC Air quality index and accuracy do not change

akirsch76, 2018-05-01:
Hello,

I'm using the bme680 Sensor and the BSEC Library for some time now and I have some problems with it.

1. The Config Files of the current release are 38 bytes long but the library only reads 34 bytes of it. I found a trick to skip the first 4 bytes. That worked but I'm not sure if this is OK. If I don't skip the first 4 bytes a get a configuration error.

2. The Air quality index and accuracy do not change even after weeks. I know that there is a calibration phase for this but nothing changes. The raw data is changing a lot over time in the correct 3s interval.

3. The sensor state files do not change over time. They are written but the content does not change.

4. The microsecond timestamps should be ok. I checked them with an online tool. But the callback function is for microsecond timestamps and in the output of the library a got a nanosecond timestamp with same value and 000 at the end (microsecond timestamp * 1000) . This could be a problem.

Can you help me with this?

Kind regards

Alex

BoschSensortec, 2018-05-02:
Hi @akirsch76,

 > The Config Files of the current release are 38 bytes long but the library only reads 34 bytes of it. I found a trick to skip the first 4 bytes. That worked but I'm not sure if this is OK. If I don't skip the first 4 bytes a get a configuration error.

The first 4 bytes contain the version number. You must be using the config files of a different version.

> The Air quality index and accuracy do not change even after weeks. I know that there is a calibration phase for this but nothing changes. The raw data is changing a lot over time in the correct 3s interval.

Does it always remain 0? This could also explain the state files.

> The microsecond timestamps should be ok. I checked them with an online tool. But the callback function is for microsecond timestamps and in the output of the library a got a nanosecond timestamp with same value and 000 at the end (microsecond timestamp * 1000) . This could be a problem.

Can you clarify why nanosecond times could be a problem?

akirsch76, 2018-05-02:
Hello,

> The first 4 bytes contain the version number. You must be using the config files of a different version.

I only have one version. The Version if got from the zip File that I've downloaded from the Bosch website. The Size of the File is 308 bytes but the library only reads 304 bytes of it.

>Does it always remain 0? This could also explain the state files.

It starts with 0. After a few minutes it raises to 1 and stays there.

>Can you clarify why nanosecond times could be a problem?

This is not a problem. I was just wondering why the library and the example want to have a microsecond timestamp from a callback routine when it uses nanosecond timestamps internally.

BoschSensortec, 2018-05-03:
@akirsch76,

We have a new repository for the BSEC library [here](https://github.com/BoschSensortec/BSEC-Arduino-library). 

- Which version of the library are you using?
- The air quality stays to one because it has only observed the same quality of air it is exposed to. In order to be more accurate, it needs to experience other air. You could for example breathe on the sensor, or expose it to VOCs, like a whiteboard marker.

serkirsch, 2018-05-03:
> We have a new repository for the BSEC library

Yes, but I need a version for the raspberry pi. But I can see that the config length in thee c++ file is also 304 Bytes instead of 308 bytes.

> Which version of the library are you using?

Version 1.4.6.0 Generic Release 20180425 that I found [here](https://www.bosch-sensortec.com/bst/products/all_products/bsec)

> The air quality stays to one because it has only observed the same quality of air it is exposed to. In order to be more accurate, it needs to experience other air. You could for example breathe on the sensor, or expose it to VOCs, like a whiteboard marker.

I tried this already but may be not long enough.