BoschSensortec/BME680_driver #52: BSEC outputting the same AQI

mekkaoui, 2018-05-10:
Hi All,
I am testing a new board using a BME680. I am using the BSEC library on a nordic 52840 SOC.
All seem to work very well except the air quality index output by the bosch apis (using the libalgobsec.a provided by Bosch) is always the same. Nothing seem to affect it even breathing out directly at the chip's vent. Everything else seem to be accurate (temperature, humidity and pressure) and reflect the environment.
The raw gas (Resistance) measurement  does reflect the changes in the environment but not the AQI.
The bosch routines were used as provided and used only via the intended interfaces.
To fake increased statistics I have increased the saving of the state by a factor of 1000 (save everything 30s). The result is the same a constant AQI of 25.0000 (25)
Your input is greatly appreciated.
Thanks

BoschSensortec, 2018-05-14:
Hi @mekkaoui,

- How long have you been measuring the Indoor Air Quality?
- Which mode are you using the BSEC library in?
- Which version of the algorithm are you using?

mekkaoui, 2018-05-14:
Hi All

    - How long have you been measuring the Indoor Air Quality?
    **For few days now (6-8days), but with power/reset interruptions. The longest it has run continuously is about 36 hours.**
    - Which mode are you using the BSEC library in?
    **I am using the bsec_iot_example.c example as is. I have just filled in the blanks (i2c, delay, timestamp and output functions). The example uses forced mode.**

    - Which version of the algorithm are you using?
    **Library version: BSEC_1.4.6.0_Generic_Release_20180425**

Thanks
Abderrezak

mekkaoui, 2018-05-14:
Update: Now it looks like the AQI has started changing and reacting to environmental change. :). Now all is working, one has to fully understand how your algorithm really does work. Thanks