BoschSensortec/BME680_driver #54: state_save / state_load is not working

michapr, 2018-05-16:
Tried again to use the state savine / loading procedure.
I always get "Error while initializing BSEC library"

Have rechecked many details, not sure what can be the reason.
```
void state_save(const uint8_t *state_buffer, uint32_t length)
{
...
}
```
Will save 65 bytes (lenght - not sure - all the time or not) to EEPROM
```
uint32_t state_load(uint8_t *state_buffer, uint32_t n_buffer)
{
}
```
Will read the same 65 bytes fro EEPROM and return the uint32_t value 65 - as described.

Compared the written values and the read values - they are the same.

Values:
`00 00 00 00 00 00 00 29 00 00 00 14 00 01 00 00 23 37 B9 60 FF 14 40 00 00 00 00 00 00 00 00 0C 00 02 00 58 0A 99 42 AA 05 3E 42 09 00 05 00 00 00 00 00 01 AA 3E 16 04 1A 5C 00 00 AA 3E 17 00 00 `

Any idea what can be wrong?

Thanks
Michael

BoschSensortec, 2018-05-16:
Hi @michapr,

Have you tried the library in the link below?

https://github.com/BoschSensortec/BSEC-Arduino-library

michapr, 2018-05-16:
tried different - "old" method (same as with 1.4.5.1) and new one, no difference.
Using now 1.4.6.0. lib.

michapr, 2018-05-16:
ok,.. understand now what you meae - the examples in BSEC-Arduino-library.
They are completely different, right.
There is working the save procedure. Anyway it original lib / functions it is not working.

In Arduino lib is a small mistake:
```
void updateState(void)
{
  bool update = false;
  if (stateUpdateCounter == 0) {
    /* First state update when IAQ accuracy is >= 1 */
    if (iaqSensor.iaqAccuracy >= 3) {
      update = true;
```
should be "**if (iaqSensor.iaqAccuracy >= 1)**" if I think right...

And, after loading the values the accuracy will be set again to "0" (was "1" before) - is this correct?
Values are not starting a long time from 25, what is better as before - but not sure about the saved calibration.

BoschSensortec, 2018-05-17:
@michapr,

As you rightly pointed out. It should be 1 here. Thanks!

miseryofdevice, 2019-02-03:
Hello,
I am facing this issue right now on BSEC library version 1.4.7.1. I am running the BME680 via a ESP8266 with Ardiuno. Code is basically the example from https://github.com/BoschSensortec/BSEC-Arduino-library/tree/master/examples/basic_config_state .
The devices tells me to store 
`0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 28 0 1 0 E7 21 B6 40 B8 22 B6 40 3C DF B1 40 26 DA B1 40 0 0 0 0 2C 1 7 7 0 0 0 40 0 0 0 40 0 0 3 0 0 0 0 0 0 0 0 0 0`
But the IAQ accuracy still resets with `0`.

```
booting up ESP8266 ...
BSEC library version 1.4.7.1
Erasing EEPROM
Timestamp [ms], raw temperature [°C], pressure [hPa], raw relative humidity [%], gas [Ohm], IAQ, IAQ accuracy, temperature [°C], relative humidity [%]
...
7343, 24.00, 95004.31, 40.01, 74891.80, 25.00, 0, 24.00, 40.01
10342, 24.03, 95004.57, 40.37, 79513.53, 25.00, 0, 23.96, 40.48
...
```

miseryofdevice, 2019-02-04:
Hi @BoschSensortec,
would be great if you can review this. We recently choose the BME680 for your next project and the sleep/reset issue is kind of a major blocker.

BoschSensortec, 2019-02-04:
Hi @miseryofdevice,

We are actively looking into this issue.

miseryofdevice, 2019-02-12:
@BoschSensortec Any updates so far? At least on a time line?

BoschSensortec, 2019-02-18:
Hi @miseryofdevice,

So far the issue reproduced only on one Arduino IDE and library installation. We haven't yet uncovered the dependency with the ESP SDK.

miseryofdevice, 2019-02-18:
In our current tool chain we have platformIO using the BSEC-Arduino-library. 
Versions:
* ESP8266 Core version: commit [3745054](https://github.com/PhracturedBlue/ESP8266MQTTMesh/commit/3745054f94f698555e8421334addeead1cdf8fbe)
* PlatformIO: version 4.0.0a3
* /.platformio/platforms/espressif8266 commit: 7b360716161fa31175ba9faca7f4e7d6960a584c

We would need a fix soon, otherwise we would need to look for an alternative sensor (which we do not want).

(Our IDE is Eclipse.)