BoschSensortec/BME680_driver #61: Humidity registers are giving always reset data

MahiThalor, 2018-07-07:
Hi 
I am able to read temperature and pressure but humidity and gas resistance registers are giving always same data* ( reset value [ 0x80 and 0x00]). i am using I2C connection.

BoschSensortec, 2018-07-09:
Hi @MahiThalor,

Can you describe your setup? 
- Which platform and IDE are you using,
- Which library are you using to control the sensor
- Snippets or a link to your application code

MahiThalor, 2018-07-09:
i am using bosch API ,from github.
Now i am able to read humidity also by changing some configuration into
set_configuration API but i am not able to read gas data.

MahiThalor, 2018-07-09:
i am not able to read gas registers. i am trying to read "gas_r_lsb" register for gas status but it is returning every time 0x04 .
Note: for gas configurations , i am setting registers as in given order:
first i am setting ,pressure ,temperate ,humidity and filter resistors and i am able to read these 3 values.
step 1) write (0x64 ,0x65) Gas_wait_0 = 0x64
step 2) write (0x5A ,0x71) Res_heat_0 = 0x5A
step 3) write (0x71 ,0x40 ) Ctr_gas_1 = 0x71