BoschSensortec/BME680_driver #66: IAQ index question

fpier, 2018-07-17:
Hi,
using a BME680 device with "lite library", version BSEC_1.4.6.0_Generic_Release_20180425, acquisition each 300 seconds, I noticed that the IAQ accuracy changes from 1 to 3 as I exptected, but sometime it comes back to 1.
Is it a normal behaviour?

[image]

Thanks
Franco

fpier, 2018-07-23:
Thanks for your comment.
Could I have some details on how works this index?
For example, from some days (at least three) the index seems to be locked to 1, although there are variations in the enviroment.
In this case, does the IAQ reading of the sensor continue to be significant?
Subsequently I attach another chart:

[image]

fpier, 2018-08-02:
Hi,
After several days of continuosly functioning, the IAQ index seems to be stucked to 1.

Please take a look to the chart I enclose:

[image]

Is this a normal behaviour?
The IAQ measurements are reliable?

Regards,
Franco

BoschSensortec, 2018-08-02:
Hi @fpier,

What kind of environment is the sensor in?

fpier, 2018-08-02:
The sensor is in our office (about 5x5 m, 4 persons working in, in office hours, there is an air condition system and sometimes we open windows).

fpier, 2018-08-28:
Hi, have you any comment to my question?
Thanks
Franco