BoschSensortec/BME680_driver #67: Change supply voltage

git4me2know, 2018-07-17:
Hello together.

I’m using the BME680 sensor connected to an Atmel SAMD20 Xplained Pro Board. Power Supply is 3.3V and the interface is 4 wire SPI.

I have done my first experience with only the basic API module files. (bme680.c, bme680.h, bme680_defs.h). The Sensor works fine and I can measure the basic sensor values like temperature, humidity or pressure. I have also a weather station near be the sensor and what fine was to see is, that this weather station has nearly the same values. (the weather station is not so bad as I thought :)

After that I wanted to implement the IAQ value for the air quality. I have taken a look to the Bosch BSEC Software (BSEC_1.4.6.0_Generic_Release_20180425). Now are starting my problems. In this .zip file I have found a PDF documentation (BST-BME680-AN008-46.pdf). In this documentation on page 6, I have found the sub chapter 1.2.2 BSEC Configuration Settings. There is written that I can change the supply voltage. But it is not explained how I can do this. This is not clear for me. I know that in the .zip file is a folder called "config". I have copied these two files “bsec_serialized_configurations_iaq.c” and “bsec_serialized_configurations_iaq.h” from the folder “generic_33v_3s_4d”. I have also include the header “bsec_serialized_configurations_iaq.h” in the module “bsec_iot_example.c” and also implemented the function “config_load”.
From my perspective it is not clearly explained what I have to do to change the different generic configurations. In the documentation is written that after the “bsec_init” the default setting “18v300s_4d” is active. How does this happen?
Another problem is now that the temperature is 10° lower than my weather station. As I have written at the beginning the value were nearly the same before I started to implement the IAQ functionality. The sensor values update every 3 seconds.
I have also the problem that the IAQ acc value is after 5 minutes stable to “1” and the IAQ value doesn’t change from the default value 25.00. As I have sometimes read in the Internet, I have tested with a whiteboard marker to trigger the value with some bad air but this doesn’t change the result. I have also waited for three days to check if the value changed.

Can maybe someone help me? Does someone have an answer for my questions?

Best regards,
Marcel (,")

BoschSensortec, 2018-07-23:
Hi @git4me2know,

- The power supply to the sensor at VDD is controlled by your hardware. If you were powering the sensor with 1.8V, the power consumed by the sensor varies. This is modeled within the BSEC software library and you need to configure it with the appropriate configuration file.
- We haven't seen cases where the temperature is lower than expected. Can you describe a bit more about the sensor breakout that you are using, and the weather station you are using as reference?
- You only have to wait for 5 minutes before it calibrates. Assuming that the software library is running and no state files are loaded in between, a marker should provide a change in the IAQ values. If not, check if the gas resistance values supplied to the algorithm are changing.

git4me2know, 2018-07-23:
Hello and thank you for your answer.

OK, I have understood that the BME680 controls the value from the supply voltage itself. I use the BME680 sensor with 3.3 volt so I have to do nothing.
I use the BME680-Breakout from Watterott electronic. All the values I’m talking about are printed out in the function “output_ready()” onto a HyperTerminal program.

Regarding the weather station it is an unspectacular device from “www.wetterdirekt.com”. I didn’t want to say that this weather station dos measure the correct temperature but I have also an alarm clock what shows a temperature and this temperature is also nearly the same value as the weather station. So it seems to be an acceptable temperature value as reference.

I have done some measurements:
The Gas value was on start up about 73.000. It takes about 5 minutes till the IAQ acc value changed to “1” and then the gas value was about 112.000. Then I hold a whiteboard marker in front of the sensor and the gas value falls down to around 4.000 (after 20 seconds). After that I have removed the whiteboard marker. Then it takes about 10 minutes till the gas value was at 85.000. Further it takes about 35 minutes till the value is back to the 112.000. After 1,5 hour the gas value comes up to 135.000.

But something was confused for me. When I implement the function “config_load()”, were I use the module and Header “bsec_serialized_configurations_iaq” (bsec_config_iaq variable). Then the temperature is 10° under the expected temperature (what I have written in my first post). But when I comment out the content from the function “config_load()” and return only zero to the function call, then the temperature is about 5° under that expected temperature. BTW the BME680 module is wired with 15cm long cables to the Atmel explained board. So the module does not have a temperature influence from something. In this case the value from gas is much higher. It is about 170.000 on start up. When the IAQ acc value changed after 5 Minutes to “1” the gas value is now 210.000. In front of the whiteboard marker the value falls also down to nearly 4.100.

But what I have also written in my first post, I have also loaded a project into my controller where the BME680 is read out with only the basic bme680 modules without the BSEC library and there is the temperature nearly the same as the weather station has. In this constellation the gas value is about 16.400.

I still do not fully understand what the function "config_load ()" is for, or the variable "bsec_config_iaq"

Thanks a lot for your patience!

Best regards,
Marcel

git4me2know, 2018-07-24:
Hello,

Sorry but for me is not so exactly clear what you mean with sample rate.

Do you mean the while loop in the function “bsec_iot_loop()”?
I have measured with a µController toggle pin in this loop and an oscilloscope. The while loop triggers every 3 seconds.

But I have found something what have changed my displayed values on my HyperTerminal program. :)
I have used a text file compare software tool to ensure that I did not have changed something important based on the original files from the BSEC zip file. And I actually found something.
In the header file “bme680_defs.h” you can find this lines:

/** BME680 configuration macros */
/** Enable or un-comment the macro to provide floating point data output */
#ifndef BME680_FLOAT_POINT_COMPENSATION
/* #define BME680_FLOAT_POINT_COMPENSATION */
#endif

I have uncommented the "#define" row so that the definition “#define BME680_FLOAT_POINT_COMPENSATION” was active. I have used the compensation in my first BME680 project. This definition seams to lead to my strange behavior.
I have commented out again the definition and then temperature was perfect. Also after 5 minute the IAQ value begins to change from 25.00 to around 35.00. But the IAQ acc value doesn’t change from “1”. After 10 minutes I have hold a whiteboard marker in front of the sensor and then the IAQ value rise very quickly up to 500. When I have removed the whiteboard marker the IAQ value falls again back to around 35.00. Also the IAQ acc value have after that changed to “3”

Can you tell me why the float point compensation leads to this strange behavior? Why could this be?

Once again, thank you for your patience!

Best regards,
Marcel

BoschSensortec, 2018-07-25:
@git4me2know,

The scaling factors are different for floating point and integer. By default the API is in integer but the wrapper around the library is expecting it in floating point. Hence the mismatch.

Thanks for pointing that out.

git4me2know, 2018-07-25:
Thank you for your efforts to help me to fix this issue :o)

closed 2018-07-25 by git4me2know