BoschSensortec/BME680_driver #68: Can we measure tier pressure??

MahiThalor, 2018-07-18:
Hi all
i did some experiments using BME680 sensor. i kept pressure into tier tube and i blew air but pressure is not changing even 50 pascal. should i change oversampling rate of pressure to measure tier pressure?
I checked pressure from ground floor to 7th floor ,there , i found 400 pascal difference is pressure. is it correct ?

BoschSensortec, 2018-07-23:
Hi @MahiThalor,

- Assuming you are trying to measure the pressure within a tire, the typical pressures within a tire are around 30 PSI which is around 2 atm which double the rated range for the BME680. In order to see a change in the pressure within the tire, you would have to exceed the pressure of the surrounding environment which is usually around 14.7 PSI at sea level and the rated range is around 15.9 PSI. 
- You can check out this [link](https://www.mide.com/pages/air-pressure-at-altitude-calculator) to check the height, but around 400 Pascals is plausible. It is roughly 36m.

MahiThalor, 2018-07-24:
Thanks @BoschSensortec for replying
Is Bosch providing any bare sensor that we can use for tier pressure measurement?

BoschSensortec, 2018-07-24:
Hi @MahiThalor,

The Bosch Sensortec product portfolio is directed towards consumer electronics. Regarding tire pressure measurement, there are no sensors currently on the market from Bosch that provide this capability.