BoschSensortec/BME680_driver #70: what are the valid range of gas residence and how to calculate IAQ?

MahiThalor, 2018-07-23:
Hi all
I am getting 1000-3000 k ohm residence. is is valid? 
I tried with 3 different-different sensor. All are giving different- different values. 
1st sensor : 200-900 k ohm 
2nd sensor : 1000-3000 k ohm
3rd sensor : 1500-3200 k ohm
*****************************************
I am heating only 1 plate ( 320 c ,150 ms)
How to calculate IAQ ?
*****************************************
I tried with IAR Cortex-M0 library for IAQ but i am getting error:
" Error[Li005]: no definition for "__iar_FSin" [referenced from HumidityTemperatureCorrector.o(libalgobsec.a)] 
Error[Li005]: no definition for "__iar_FLog" [referenced from IaqEstimator.o(libalgobsec.a)]
"
******************************************************
 For IAR Cortex-M0 library , where is the flies "bsec_datatypes.h" and "bsec_interface" .
These files path should be "BSEC_1.4.6.0_Generic_Release_20180425\algo\bin\Lite_version\IAR\cortex_m0" , but here, only library is present.
***********************************************
Please reply asap.
Thanks and Regards
Mahendra

juergs, 2019-01-01:
I'm not agreeing with 523p.
We have some working implementations using ESP8266 or Maple-Mini boards.
BSEC implementation is documented here from "Wolfgang Klenk" with a working BSEC-Lib configuration.
The sensor needs usually 4 days as burn-in phase then works in my opinion great and sensitive.
[hint](http://bfy.tw/La9l)
regrads,
Juergen

Kormykake, 2019-01-25:
I'am also interested to know what the gas resistance values can be considered "normal". My sensor outputs values in mega ohms like 3 958 641 with the parameters of 350 C and duration=150 ms. Temperature, Humidity and Pressure values are reasonable.  (Currently, I'am hunting the programming error from the driver code. I have heavily changed the driver code for my platform but functionally I have tried to stick to that how the driver initializes and triggers bme680 in Bosch's original code). I have also wondered few calibration values. Can they be 0:
calib.res_heat_range = 0
calib.res_heat_val = 0
calib.res_sw_err =0

The other odd thing is the the heater stability bit (heat_stab_r at 0x2b) that don't go "high-state" after measurement.