BoschSensortec/BME680_driver #71: SPI issue using MBED

xuio, 2018-07-25:
I want to use the BME680 Sensor on a custom board connected to an NRF52832, which is programmed using MBED.

These are my transfer functions:
```C++
inline void user_delay_ms(uint32_t period){
	wait_ms(period);
}

int8_t user_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len){
	spi_cs=0;

	spi.write(reinterpret_cast<const char *>(&reg_addr), 1, reinterpret_cast<char *>(reg_data), len);

	spi_cs=1;

	int8_t rslt = 0;

	return rslt;
}

int8_t user_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len){
	spi_cs=0;

	uint8_t recv_buffer = 0;

	uint8_t write_buffer[10] = {0};

	write_buffer[0] = reg_addr;

	for(uint8_t i = 0; i < len; i++){
		write_buffer[i+1] = reg_data[i];
	}

	spi.write(reinterpret_cast<const char *>(write_buffer), 1 + len, reinterpret_cast<char *>(&recv_buffer), 0);

	spi_cs=1;

    int8_t rslt = 0;
    return rslt;
}
```

Do you have any idea why it doesn't work?

Thanks for your help!

~Moritz

BoschSensortec, 2018-07-26:
@xuio,

For the SPI write, you will need up to 20bytes to be written. Also, what exactly doesn't work?

xuio, 2018-07-26:
When initializing the BME680, it returns error code `BME680_E_DEV_NOT_FOUND`.
When I try to read some data from it, every value is 0:
```C++
(gdb) print bme680_get_sensor_data(&data, &bme)
$4 = 2 '\002'

(gdb) print data
$3 = {status = 0 '\000', gas_index = 0 '\000', meas_index = 0 '\000', temperature = 0,
  pressure = 0, humidity = 0, gas_resistance = 0}
```

The schematic is very simple and shouldn't be an issue:
[image]

BoschSensortec, 2018-07-27:
@xuio,

How have you configured the SPI and CS?

xuio, 2018-07-27:
These are my SPI and CS constructors:
```C++
SPI spi(P0_10, P0_12, P0_14);

DigitalOut spi_cs(P0_5);

[...]

spi.format(8, 0);
spi.frequency(1000000);

[...]

bme.dev_id = 0;
bme.intf = BME680_SPI_INTF;
bme.read = user_spi_read;
bme.write = user_spi_write;
bme.delay_ms = user_delay_ms;
```

`P0_10` normally has NFC function on NRF52832, but I configured it as GPIO using preprocessor macro `CONFIG_NFCT_PINS_AS_GPIOS`

TobiasRoeddiger, 2018-08-01:
@BoschSensortec,

Any updates on this? 😊

xuio, 2018-08-01:
@BoschSensortec 
Update:
Communication is working now, but when reading data every value is still 0, and the function call returns `BME680_W_NO_NEW_DATA`

BoschSensortec, 2018-08-01:
@xuio,

There are few reasons why this would be the case,
 - The forced mode is not triggered
 - Not sufficient time has passed after triggering the forced mode and reading data.

xuio, 2018-08-01:
@BoschSensortec 
Time should not be an issue, I tried single stepping through the debugger with sufficient delays.

How to trigger forced mode? This is my whole main function:

```C++
int main(){
	spi_cs = 0;
	spi.format(8, 0);
	spi.frequency(500000);
	uint8_t set_required_settings;

	bme.dev_id = 0;
	bme.intf = BME680_SPI_INTF;
	bme.read = user_spi_read;
	bme.write = user_spi_write;
	bme.delay_ms = user_delay_ms;
	bme.amb_temp = 25;
	bme.gas_sett.run_gas = BME680_ENABLE_GAS_MEAS;
	bme.gas_sett.heatr_temp = 320;
	bme.gas_sett.heatr_dur = 150;
	bme.power_mode = BME680_FORCED_MODE;

	set_required_settings = BME680_OST_SEL | BME680_OSP_SEL | BME680_OSH_SEL | BME680_FILTER_SEL | BME680_GAS_SENSOR_SEL;

	bme680_set_sensor_settings(set_required_settings,&bme);
	bme680_set_sensor_mode(&bme);

	// open bme
	int8_t rslt = BME680_OK;
	rslt = bme680_init(&bme);
	if(rslt != BME680_OK){
		// handle error
		if(rslt == BME680_E_DEV_NOT_FOUND){
			uint8_t t = 0;
		}
	}else{
		//queue.call_every(400, updateSensor);
	}

	while (1){
		uint16_t meas_period;
		bme680_get_profile_dur(&meas_period, &bme);
		wait_ms(meas_period);

		int8_t rslt = bme680_get_sensor_data(&data, &bme);
		if(rslt = BME680_W_NO_NEW_DATA){
			int t=0;
		}
	}
}
```