BoschSensortec/BME680_driver #72: BSEC linking fails.

wouterjoris, 2018-07-29:
Hi there, 

Not sure if this is the correct spot to ask this question but I see there have been others here with similar questions.

When I try to build the BSEC library with KEIL for a NXP LPX1114 device, I get these errors. 

 `linking...
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_get_state (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_set_configuration (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_set_state (referred from de1801-bme680.o).
Not enough information to list image symbols.
Finished: 1 information, 0 warning and 3 error messages.`

When I remove the libalglobsec.lib all together I get much more errors 

`linking...
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_do_steps (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_get_state (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_init (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_sensor_control (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_set_configuration (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_set_state (referred from de1801-bme680.o).
.\Objects\DE1801A.axf: Error: L6218E: Undefined symbol bsec_update_subscription (referred from de1801-bme680.o).
Not enough information to list image symbols.`

Which gives me the impression the precompiled lib is being found but there is another issue present. 
Is there anyone with the same experiences? or some tips, any help here is welcome.

BoschSensortec, 2018-07-30:
@wouterjoris,

- Which version of the BSEC library are you using?
- What is the whole link command?

Darcy123, 2018-07-31:
Hello @wouterjoris ,

Please comment 'bsec_get_state', 'bsec_set_state' and 'bsec_set_configuration' in 'bsec_integration.c', you are using lite version and those three functions are not included in the lite version lib.

wouterjoris, 2018-07-31:
@Darcy123 

Thanks for your reply. I was not aware these functions were omitted in the lite version. I must heve missed this in the appication note.

Thanks for the clarification.

closed 2018-07-31 by wouterjoris