BoschSensortec/BME680_driver #80: sensor reading examples?

skitsanos, 2018-11-16:
Not really an issue, but rather a question.

Are there any complete samples on how to read data from the sensor? The ones that were provided with BME280 codebase working just fine for me, but on BME680 it seems i'm stuck on I2C write and read handlers...

on a side note, i'm on freertos..

Thank you.

skitsanos, 2018-12-17:
Seems i'm doing something wrong with i2c read write...

```
esp_err_t bosch_i2c_master_init(i2c_mode_t mode, gpio_num_t sda_io_num, gpio_num_t scl_io_num)
{
    i2c_config_t i2c_config = {
            .mode = mode,
            .sda_io_num = sda_io_num,
            .scl_io_num = scl_io_num,
            .sda_pullup_en = GPIO_PULLUP_ENABLE,
            .scl_pullup_en = GPIO_PULLUP_ENABLE,
            .master.clk_speed = 1000000
    };

    i2c_param_config(I2C_NUM_0, &i2c_config);

    esp_err_t ret = i2c_driver_install(I2C_NUM_0, mode, 0, 0, 0);
    if (ret == ESP_OK)
    {
        ESP_LOGI(TAG, "I2C driver installed. SDA:%d, SCL:%d", sda_io_num, scl_io_num);
    }
    else
    {
        ESP_LOGI(TAG, "I2C driver failed. Err code: %d", ret);
    }

    return ret;
}

int8_t bosch_i2c_bus_write(uint8_t addr, uint8_t reg, uint8_t *data, uint16_t len)
{
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, addr << 1 | I2C_MASTER_WRITE, true);
    if (reg)
    {
        i2c_master_write_byte(cmd, reg, true);
    }
    if (data)
    {
        i2c_master_write(cmd, data, len, true);
    }
    i2c_master_stop(cmd);
    esp_err_t err = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    ESP_LOGW(TAG, "bosch_i2c_bus_write: %d",err);

    return err;
}

int8_t bosch_i2c_bus_read(uint8_t bus, uint8_t addr, const uint8_t reg, uint8_t *data, uint16_t len)
{
    if (len == 0)
    { return true; }

    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    if (reg)
    {
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, (addr << 1) | I2C_MASTER_WRITE, true);
        i2c_master_write_byte(cmd, reg, true);
        if (!data)
        {
            i2c_master_stop(cmd);
        }
    }
    if (data)
    {
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, (addr << 1) | I2C_MASTER_READ, true);
        if (len > 1)
        { i2c_master_read(cmd, data, len - 1, I2C_ACK_VAL); }
        i2c_master_read_byte(cmd, data + len - 1, I2C_NACK_VAL);
        i2c_master_stop(cmd);
    }
    esp_err_t err = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    ESP_LOGW(TAG, "bosch_i2c_bus_read: %d",err);

    return err;
}
```

It passes the i2c init just fine, then when i do bme680 with my config, it all goes nuts.

```
struct bme680_dev gas_sensor;
    gas_sensor.dev_id = BME680_I2C_ADDR_PRIMARY;
    gas_sensor.intf = BME680_I2C_INTF;
    gas_sensor.read = bosch_i2c_bus_read;
    gas_sensor.write = bosch_i2c_bus_write;
    gas_sensor.delay_ms = bosch_delay_msek;
    gas_sensor.amb_temp = 25;

int8_t rslt = bme680_init(&gas_sensor);
    if (rslt != BME680_OK)
    {
        ESP_LOGE(TAG, "Failed to init BME680");
    }
    else
    {
        ESP_LOGI(TAG, "BME680 init OK");
    }
```

aureq, 2019-02-08:
@skitsanos Did you fix your issue?

skitsanos, 2019-02-15:
@aureq somewhat, not in the way i wanted.... for the moment i use https://github.com/gschorcht/bme680-esp-idf I really wanted to have clean code without wrappers, to be it like Bosch provided BME280 sample... In my case every byte counts...

You can see my BME680 spitting data to [http://view.loggerdaisy.com/devices/24ac472644](http://view.loggerdaisy.com/devices/24ac472644) I collect raw data, because Air Quality Index is not relevant for my project, i hunt other ghosts...

aureq, 2019-02-16:
@skitsanos That's pretty cool. What do you use to get the TVOX(ppb) ?

skitsanos, 2019-02-18:
@aureq for the moment i use combo of BME680 + CCS811 (a tricky one btw), but i need to add SGP30 to my set, unfortunately can't find any implementation of the driver for it and my C skills are not the same as other languages, although i'm doing crazy job here... So i really need to solve that SGP30 addition...