BoschSensortec/BME680_driver #81: Not measuring CO2 and bVoc

rado99, 2018-12-11:
Hello, I am using and ESP32 board and I2C interface, with your example code "bsec_iot_example". Everything is working nice, except these two values. Can you give any suggestions?

Ovidius80, 2019-01-15:
you will have to modify the bsec_integration.c

after 

`
                case BSEC_OUTPUT_STABILIZATION_STATUS:
                    gas_percentage = bsec_outputs[index].signal;
                    gas_percentage_acccuracy = bsec_outputs[index].accuracy;
                    break;`

add

`
                case BSEC_OUTPUT_CO2_EQUIVALENT:
                    co2_equivalent = bsec_outputs[index].signal;
                    co2_accuracy = bsec_outputs[index].accuracy;
                    break;
                case BSEC_OUTPUT_BREATH_VOC_EQUIVALENT:
                    breath_voc_equivalent = bsec_outputs[index].signal;
                    breath_voc_accuracy = bsec_outputs[index].accuracy;
                    break;`

now you can use it in the function void output_ready of your bsec_iot_example.

According to the libray both values are in ppm. CO2 starts with 500ppm and VOC with 0,500 ppm. 
VOC seems to go up to 1000. 
An other sensor from a different compay i have, measures VOC in ppb and the Values are way different.

rado99, 2019-01-15:
Thank you. I am using the Seeed version of the sensor, but with the original examples and libraries. Does that mean I can expect only ppm, without expecting ppb values?

rado99, 2019-01-21:
It is interesting that, after adding these lines there is no change.

Ovidius80, 2019-01-22:
Well for me it works. just be sure to use the newest bsec library version