BoschSensortec/BME680_driver #82: BSEC lite GCC CortexM0 taking ~32KB instead of the ~13KB stated

damancusonqn, 2018-12-26:
Hi,
I'm trying to integrate the BSEC in a STM32F072C8U6 (64KB flash).
My application (including the BME680 driver) is using 41KB. The BSEC_Binary_Size_Information.pdf doc states 13KB for the GCC-CortexM0 version of algobsec. But when I link it the size of the binary goes to ~73KB (so it won't fit in flash).
Analyzing the generated MAP file I see that effectively the algobsec objects are using around 13KB, but the libgcc.a is taking ~5KB extra and libm.a (I'm not using it in my app) another ~14KBs.

Is this expected? Is anything that I can do to reduce the size?
I'm already using optimizations (-Os) and removed the debug information. I'm using newlib_nano.

I'm adding some compile log in case it helps:

------------------------------
Not using BSEC (including in makefile but not called from app):
``` 
/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi-gcc -c -mcpu=cortex-m0  -march=armv6-m -mthumb  -mfloat-abi=soft -DUSE_HAL_DRIVER -DSTM32F072xB -DAPP_DEBUG_LEVEL=0 -IInc -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/BME680_driver -IMiddlewares/Bosch -Os -Wall -fdata-sections -ffunction-sections  -fmessage-length=0 -fno-exceptions -fno-builtin -fomit-frame-pointer -fuse-linker-plugin  -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Src/main.c -o build/main.o
CC Src/main.c
/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi-gcc build/main.o build/usb_device.o build/usbd_conf.o build/usbd_desc.o build/usbd_cdc_if.o build/stm32f0xx_it.o build/stm32f0xx_hal_msp.o build/stm32f0xx_hal_pcd.o build/stm32f0xx_hal_pcd_ex.o build/stm32f0xx_hal_i2c.o build/stm32f0xx_hal_i2c_ex.o build/stm32f0xx_hal_tim.o build/stm32f0xx_hal_tim_ex.o build/stm32f0xx_hal_rcc.o build/stm32f0xx_hal_rcc_ex.o build/stm32f0xx_hal.o build/stm32f0xx_hal_gpio.o build/stm32f0xx_hal_dma.o build/stm32f0xx_hal_cortex.o build/stm32f0xx_hal_pwr.o build/stm32f0xx_hal_pwr_ex.o build/stm32f0xx_hal_flash.o build/stm32f0xx_hal_flash_ex.o build/system_stm32f0xx.o build/usbd_core.o build/usbd_ctlreq.o build/usbd_ioreq.o build/usbd_cdc.o build/stm32f0xx_hal_uart.o build/stm32f0xx_hal_uart_ex.o build/bme680.o build/syscalls.o build/thConfig.o build/thBsec.o build/startup_stm32f072xb.o -mcpu=cortex-m0  -march=armv6-m -mthumb  -mfloat-abi=soft  -specs=nano.specs -u _printf_float -TSTM32F072C8Ux_FLASH.ld -L Middlewares/Bosch -lc -lalgobsec -lm -lnosys   -Wl,-Map=build/USBthingVOC.map,--cref -Wl,--gc-sections -o build/USBthingVOC.elf
Linking... build/main.o
-------------------------------------------------------------------------------
   text	   data	    bss	    dec	    hex	filename
  42104	    824	   4312	  47240	   b888	build/USBthingVOC.elf
-------------------------------------------------------------------------------
```
Calling  bsec_get_version() for instance:

```
With bsec:
-------------------------------------------------------------------------------
   text	   data	    bss	    dec	    hex	filename
  75016	    828	   5400	  81244	  13d5c	build/USBthingVOC.elf
-------------------------------------------------------------------------------
```
And the MAP files for both cases:
[output.map.bsec.txt](https://github.com/BoschSensortec/BME680_driver/files/2710226/output.map.bsec.txt)
[output.map.noBsec.txt](https://github.com/BoschSensortec/BME680_driver/files/2710227/output.map.noBsec.txt)

Thanks in advance!!
All the best,
Daniel Mancuso.

BoschSensortec, 2019-02-01:
Hi @damancusonqn,

Yes this is expected as the BSEC library depends on the math library.

damancusonqn, 2019-02-01:
OK, thank you.
Closing the issue then.

Best.

closed 2019-02-01 by damancusonqn