BoschSensortec/BME680_driver #84: Selftest: Minimum humidity limit too high?

ojousima, 2019-01-21:
The self-test of BME680 relies on actual indoor humidity being over 20% RH. While this is a valid assumption most of the time, for example right now the temperature is a bit below -20C in Helsinki, Finland. This leads to dry indoor air (18 % in my case) which in turn causes self-test to fail.

BoschSensortec, 2019-02-01:
Hi @ojousima,

Yes, you are right. The self-test snippet is a reference for the end-of-line production test of the devices using the BME680. Our existing customers run this indoors at room temperature. You would have to modify the self-test to suit your test environment.

ojousima, 2019-02-01:
Thanks,

Just to clarify, we're also running this indoors at room temperature. However the outdoor conditions (extremely dry air) are reflected to indoors. 

I'll adjust the test on our end and close this issue.

closed 2019-02-01 by ojousima